raulmur/ORB_SLAM2 #678: How to convert the sparse map from ORB_SLAM2 to a navigation map

ArlenCHEN, 2018-11-06:
Hi all,

Firstly thank you very much, Raul, for sharing your work. I can get a good sparse mapping result.
However, what I am interested in is to use the map to navigation, the sparse map does not provide detailed information about obstacles and there are also a lot of noisy points.
Anyone has some ideas how to convert the sparse map to a navigatable one so that we could represent the map using octomap and further we can implement navigation in the map.

BTW, is there any difference between the map from monocular mode and the map from RGBD mode, I can see both maps are sparse one. Any idea about this point?

Thanks a lot!!

a-jahani, 2018-11-08:
Hi,

This might be useful to you:
[2D grid mapping and navigation with ORB-SLAM 2- KITTI Dataset
](https://www.youtube.com/watch?v=FCd6p25131I&t=44s
)

We've been projected orbslam map points into a 2D grid map through a course project. Although the points were still sparse but might give you the initial idea of how to start.

Best,
Ali

ArlenCHEN, 2018-11-09:
Hi Ali,

Nice work! Thanks a lot for your information!
Zheng

govindak-umd, 2021-03-05:
Did you actually try to run what was in the video?  As per that link, In pointCloudToGridMap2d.py, I changed the keyframe_trajectory_fname to read KeyFrameTrajectory.txt. What do I read as point_cloud_fname?

chunibyo-wly, 2021-03-18:
What am I do is write some code to expose keyframe and pointclouds in this keyframe in one file.
reference this [link](https://github.com/chunibyo-wly/ORB_SLAM3_MOKAI/commit/fa714246ccf87fb15f0933f1a59130498447ce56).