raulmur/ORB_SLAM2 #679: Issue reegarding building for including some libpng16.0.16 file

DikshitDHegde, 2018-11-07:


LongruiDong, 2019-03-13:
> /home/dikshit/anaconda3/lib/libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'
> collect2: error: ld returned 1 exit status
> 
> this is the error so what should i do
Hi, I met the same error ,Have you figured it out ?
Thanks!

KaninchenM, 2020-07-27:
> > /home/dikshit/anaconda3/lib/libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'
> > collect2: error: ld returned 1 exit status
> > this is the error so what should i do
> 
> Hi, I met the same error ,Have you figured it out ?
> Thanks!

Have you solved it ? I meet the same problem. Maybe the wrong use of `Anaconda`?

KaninchenM, 2020-07-27:
> Have you solved it ? I meet the same problem. Maybe the wrong use of `Anaconda`?

It may be caused by the version of ZLIB.  It seems the `libpng16.so.16` suitable for `ZLIB1.2.8` but not suitable for `ZLIB1.2.11`