raulmur/ORB_SLAM2 #681: How could I compare the pose result of ORB_SLAM with the KITTI ground truth?

AbnerCSZ, 2018-11-10:
How could I compare the pose result of mine algorithm(SLAM/Odometry) with the ground truth?
Just like tum rgb-d tool:
[Useful tools for the RGB-D benchmark](https://vision.in.tum.de/data/datasets/rgbd-dataset/tools)  

In other words, how could I calculate the ate(ABSOLUTE TRAJECTORY ERROR) and RELATIVE POSE ERROR(rpe) between My pose result  and the benchmark of KITTI?

Thanks!

AbnerCSZ, 2018-11-30:
> You can try with this: https://github.com/MichaelGrupp/evo
> 
> #shamelessadvertising

Thank you very much!

closed 2018-11-30 by AbnerCSZ

d3dileep, 2020-02-06:
Is APE in Evo the same as ATE for Kitti evaluation? I am using https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics_interactive.ipynb as a reference. I found an evaluate_ate.py script for the TUM dataset in the repository. What is the fundamental difference between APE and ATE?

AbnerCSZ, 2020-02-09:
> Is APE in Evo the same as ATE for Kitti evaluation? I am using https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics_interactive.ipynb as a reference. I found an evaluate_ate.py script for the TUM dataset in the repository. What is the fundamental difference between APE and ATE? @AbnerCSZ @MichaelGrupp

APE(absolute pose error) in evo includes errors in rotation and translation. You can refer to their code: 
[class APE(PE)](https://github.com/MichaelGrupp/evo/blob/3b3cf9d3f689f7236076bf8544af16a84b99f478/evo/core/metrics.py#L378)

```
        if self.pose_relation == PoseRelation.translation_part:
            # E is an array of position vectors only in this case
            self.error = [np.linalg.norm(E_i) for E_i in self.E]
        elif self.pose_relation == PoseRelation.rotation_part:
            self.error = np.array([
                np.linalg.norm(lie.so3_from_se3(E_i) - np.eye(3))
                for E_i in self.E
            ])
        elif self.pose_relation == PoseRelation.full_transformation:
            self.error = np.array(
                [np.linalg.norm(E_i - np.eye(4)) for E_i in self.E])
        elif self.pose_relation == PoseRelation.rotation_angle_rad:
            self.error = np.array(
                [abs(lie.so3_log(E_i[:3, :3])) for E_i in self.E])
        elif self.pose_relation == PoseRelation.rotation_angle_deg:
            self.error = np.array([
                abs(lie.so3_log(E_i[:3, :3])) * 180 / np.pi for E_i in self.E
            ])
        else:
            raise MetricsException("unsupported pose_relation")
```
ATE(absolute trajectory error) in KITTI only evaluates translation errors.