raulmur/ORB_SLAM2 #682: When I run the unary dataset, I can't display the window. The window just blinks and disappears.

cugcsu, 2018-11-12:
When I tested the unary data on Ubuntu 16.04:
`./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM3.yaml '/home/gsx/Downloads/rgbd_dataset_freiburg1_xyz/depth' 
`
and then occurred:
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.0.0-beta) /home/gsx/Downloads/opencv-4.0.0/modules/highgui/src/window.cpp:615: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvNamedWindow'

Aborted (core dumped)

closed 2018-11-14 by cugcsu

cisnow, 2019-09-27:
BRO,have you solved this problem?I met the same problem

jzhzhang, 2019-10-19:
@cugcsu 
hi, have you solved this problem?

DenisLim95, 2019-10-23:
@cugcsu wondering same thing... have you solved this?

the-cat-crying, 2020-01-19:
BRO,have you solved this problem?I met the same problem

biendltb, 2020-03-03:
I solved this problem by building and re-installing OpenCV version 3.4.7 from the source.
It won't work for version 4.1 and 3.2.4 as my own tests.