raulmur/ORB_SLAM2 #686: Ros  Example Running Stereo Node Online

cjcjlin, 2018-11-22:
> For a stereo input from topic /camera/left/image_raw and /camera/right/image_raw run node ORB_SLAM2/Stereo. 

How can I create the stereo topic **/camera/left/image_raw** and **/camera/right/image_raw** ?

ShettyHarapanahalli, 2019-01-15:
Does the space between the cameras have any effect on the SLAM ?

ibaranov-cp, 2019-01-15:
Yes, to a point.

Assuming everything is well calibrated, there should not be much of a difference.

However, the closer your cameras are, the shorter your baseline, and hence the less accurate you are at longer distances. (if all your features are far away, go wider)

Conversely, the farther apart your cameras are, the longer minimum range your sensor will have, as both cameras need to see a feature at the same time. (if all your features are close by, go closer)