raulmur/ORB_SLAM2 #705: Display current points on Map Viewer

PierBJX, 2019-01-08:
Using monocular, I would like to know if it was possible to color the current points that the camera sees (green square on "Current Frame" window) on the Map Viewer window.
 
[image]

ORB SLAM uses two functions to get the points on the map ([code](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/MapDrawer.cc#L46)):
```
    const vector<MapPoint *> &vpMPs = mpMap->GetAllMapPoints();          // Black points
    const vector<MapPoint *> &vpRefMPs = mpMap->GetReferenceMapPoints(); // Red points
```

From these two vectors how could I find the current points display on the camera ? 
I found that the current points displayed on the current frame window are mCurrentFrame.mvKeys. However, these points are not vector Map Point but a vector of KeyPoints.

AlejandroSilvestri, 2019-01-18:
@PierBJX 

Let say you want to populate the custom variable

     const vector<MapPoint *> &vpCurrentMPs  = SomehowGetCurrentMapPoints();          // Green points

You can retrieve those mappoints from

    (vector<MapPoint*>) mCurrentFrame.mvpMapPoints  

This vector has size `mCurrentFrame.N` and most of its elements are null.

PierBJX, 2019-01-18:
Thanks, I succeeded to do it. Here is the code to modify : 

**Map.h**

- Insert [line 54](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/include/Map.h#L54)
```
  // Function to manage the current points to color them in green
  void AddCurrentMapPoint(MapPoint *pMP);        // add
  void EraseCurrentMapPoint();                   // delete
  std::vector<MapPoint *> GetCurrentMapPoints(); // get
```
- Insert [line 72](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/include/Map.h#L72)
`   std::set<MapPoint *> mspCurrentMapPoints; // Current map points`

**MapDrawer.h**
- Change [line 41](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/include/MapDrawer.h#L41) with
`   void DrawMapPoints(const bool bDrawCurrentPoints);`

**Map.cc**
- Insert [line 132](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Map.cc#L132)
```
void Map::AddCurrentMapPoint(MapPoint *pMP)
{
    unique_lock<mutex> lock(mMutexMap);
    mspCurrentMapPoints.insert(pMP);
}

void Map::EraseCurrentMapPoint()
{
    unique_lock<mutex> lock(mMutexMap);
    mspCurrentMapPoints.clear();
}

vector<MapPoint *> Map::GetCurrentMapPoints()
{
    unique_lock<mutex> lock(mMutexMap);
    return vector<MapPoint *>(mspCurrentMapPoints.begin(), mspCurrentMapPoints.end());
}
```

- Add a line in the [clear function line 118](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Map.cc#L118)
 ` mspCurrentMapPoints.clear();`

**MapDrawer.cc**
- Change [line 44](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/MapDrawer.cc#L44) with
`void MapDrawer::DrawMapPoints(const bool bDrawCurrentPoints)`

- Insert after the [line 80](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/MapDrawer.cc#L80)
```
    if (bDrawCurrentPoints)
    {
        // Define points
        glPointSize(5);
        glBegin(GL_POINTS);
        glColor3f(0.0, 1.0, 0.0);

        // All map points
        for (std::vector<MapPoint *>::const_iterator i = vpCurrentMPs.begin(); i != vpCurrentMPs.end(); i++)
        {
            if ((*i)->isBad())
                continue;
            cv::Mat pos = (*i)->GetWorldPos();
            glVertex3f(pos.at<float>(0), pos.at<float>(1), pos.at<float>(2));
        }
        glEnd();
    }
```

**Tracking.cc**
- Insert [line 419](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Tracking.cc#L419)
```
        // Empty the current map points vector
        mpMap->EraseCurrentMapPoint();

        // Add the current map points in the vector
        for (int i = 0; i < mCurrentFrame.N; i++)
        {
            if (mCurrentFrame.mvpMapPoints[i] && !mCurrentFrame.mvbOutlier[i])
            {
                mpMap->AddCurrentMapPoint(mCurrentFrame.mvpMapPoints[i]);
            }
        }

```

**Viewer.cc**
- Change [line 68](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Viewer.cc#L68) with
`    pangolin::CreatePanel("menu").SetBounds(0.0, 1.0, 0.0, pangolin::Attach::Pix(185));`

- Insert [line 72](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Viewer.cc#L72)
`    pangolin::Var<bool> menuShowCurrentPoints("menu.Show Current Points", true, true);`

- Change [line 133](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Viewer.cc#L133) with
`            mpMapDrawer->DrawMapPoints(menuShowCurrentPoints);`

closed 2019-01-18 by PierBJX

AlejandroSilvestri, 2019-01-19:
@PierBJX , glad to know you succeeded.

If you have any time, please post a screen capture, and better if you can upload a video to YouTube and share the link here with us.

PierBJX, 2019-01-20:
@AlejandroSilvestri, I have no time yet to do it but I will and share the link.

rahulsharma11, 2019-02-08:
Hi @PierBJX ,
I tried exact same modifications but still getting only red and black points.
Is there any other modification left?
Thanks.

PierBJX, 2019-02-08:
@rahulsharma11 
Have you built orb-slam ?

rahulsharma11, 2019-02-08:
Yes, i have built orbslam.

[image]

KaninchenM, 2020-06-28:
> Thanks, I succeeded to do it. Here is the code to modify :
> 
> **Map.h**
> 
> * Insert [line 54](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/include/Map.h#L54)
> 
> ```
>   // Function to manage the current points to color them in green
>   void AddCurrentMapPoint(MapPoint *pMP);        // add
>   void EraseCurrentMapPoint();                   // delete
>   std::vector<MapPoint *> GetCurrentMapPoints(); // get
> ```
> 
> * Insert [line 72](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/include/Map.h#L72)
>   `   std::set<MapPoint *> mspCurrentMapPoints; // Current map points`
> 
> **MapDrawer.h**
> 
> * Change [line 41](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/include/MapDrawer.h#L41) with
>   `   void DrawMapPoints(const bool bDrawCurrentPoints);`
> 
> **Map.cc**
> 
> * Insert [line 132](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Map.cc#L132)
> 
> ```
> void Map::AddCurrentMapPoint(MapPoint *pMP)
> {
>     unique_lock<mutex> lock(mMutexMap);
>     mspCurrentMapPoints.insert(pMP);
> }
> 
> void Map::EraseCurrentMapPoint()
> {
>     unique_lock<mutex> lock(mMutexMap);
>     mspCurrentMapPoints.clear();
> }
> 
> vector<MapPoint *> Map::GetCurrentMapPoints()
> {
>     unique_lock<mutex> lock(mMutexMap);
>     return vector<MapPoint *>(mspCurrentMapPoints.begin(), mspCurrentMapPoints.end());
> }
> ```
> 
> * Add in the [clear function line 118](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Map.cc#L118)
>   ` mspCurrentMapPoints.clear();`
> 
> **MapDrawer.cc**
> 
> * Change [line 44](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/MapDrawer.cc#L44) with
>   `void MapDrawer::DrawMapPoints(const bool bDrawCurrentPoints)`
> * Insert after the [line 80](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/MapDrawer.cc#L80)
> 
> ```
>     if (bDrawCurrentPoints)
>     {
>         // Define points
>         glPointSize(5);
>         glBegin(GL_POINTS);
>         glColor3f(0.0, 1.0, 0.0);
> 
>         // All map points
>         for (std::vector<MapPoint *>::const_iterator i = vpCurrentMPs.begin(); i != vpCurrentMPs.end(); i++)
>         {
>             if ((*i)->isBad())
>                 continue;
>             cv::Mat pos = (*i)->GetWorldPos();
>             glVertex3f(pos.at<float>(0), pos.at<float>(1), pos.at<float>(2));
>         }
>         glEnd();
>     }
> ```
> 
> **Tracking.cc**
> 
> * Insert [line 419](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Tracking.cc#L419)
> 
> ```
>         // Empty the current map points vector
>         mpMap->EraseCurrentMapPoint();
> 
>         // Add the current map points in the vector
>         for (int i = 0; i < mCurrentFrame.N; i++)
>         {
>             if (mCurrentFrame.mvpMapPoints[i] && !mCurrentFrame.mvbOutlier[i])
>             {
>                 mpMap->AddCurrentMapPoint(mCurrentFrame.mvpMapPoints[i]);
>             }
>         }
> ```
> 
> **Viewer.cc**
> 
> * Change [line 68](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Viewer.cc#L68) with
>   `    pangolin::CreatePanel("menu").SetBounds(0.0, 1.0, 0.0, pangolin::Attach::Pix(185));`
> * Insert [line 72](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Viewer.cc#L72)
>   `    pangolin::Var<bool> menuShowCurrentPoints("menu.Show Current Points", true, true);`
> * Change [line 133](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Viewer.cc#L133) with
>   `            mpMapDrawer->DrawMapPoints(menuShowCurrentPoints);`

Thanks to PierBJX.
Remember to add the following content to function`DrawMapPoints(const bool bDrawCurrentPoints)` in file `MapDrawer.cc`:
    `const vector<MapPoint*> &vpCurrentMPs = mpMap->GetCurrentMapPoints(); `
After `const vector<MapPoint*> &vpRefMPs = mpMap->GetReferenceMapPoints();`
Or you will not compile well.

RehabHsn, 2022-01-09:
If I want to save the map points to PCL format or any format, how can I do that?
Thanks in advance

KaninchenM, 2022-01-09:
您的邮件已收到，我会尽快回复您！

AlejandroSilvestri, 2022-01-10:
@RehabHsn 

This project let you save and load your maps:

https://github.com/AlejandroSilvestri/osmap

MapPoints go in separated files.  You can program it to save in other formats.