raulmur/ORB_SLAM2 #708: Can ORB SLAM 2 be used in a simulated environment such as AirSim

ivannson, 2019-01-10:
I was wondering if it is possible to apply ORB SLAM in AirSim, which runs in UnrealEngine4.

ShettyHarapanahalli, 2019-01-15:
Hi we can use the AIRSIM with ROS, so within the ROS environment we can use the ORB SLAM as well.

ivannson, 2019-02-06:
@ShettyHarapanahalli How did you calibrate the camera?

kritikagovindaraj, 2020-06-04:
@ShettyHarapanahalli  @ivannson Can this be implemented on AirSim without ROS? I am planning to use the PythonClient to implement it

ivannson, 2020-06-04:
@kritikagovindaraj Hi, I don't think I used ROS. My purpose was fairly simplistic, so I recorded a video from AirSim, and used it as a test input into ORB SLAM 2, but it wasn't running live. I think you should be able to do it without ROS, but using ROS might make it easier.