raulmur/ORB_SLAM2 #710: I'm using an Intel RealSense R200 camera and was wondering if it's possible to build a depth map based on color

lnoah995, 2019-01-14:


FPSychotic, 2019-01-17:
> I'm using an Intel RealSense R200 camera and was wondering if it's possible to build a depth map based on color with the RGB/IR camera available. Thanks.

how you made to make it run, i have it installend but I dont know how to run it with the r200 in mono,streo and rb-d if is possible. hank you

AlejandroSilvestri, 2019-01-18:
@lnoah995 

I did it, you may be interested in the code.

First I added a [`vector<cv::Vec3b> vRgb;`](https://github.com/AlejandroSilvestri/os1/blob/179e42d8f940530989bae24086e6a339e62f8321/include/KeyFrame.h#L670) property to KeyFrame, populated in KeyFrame's constructor, needing access to the color image to gather keypoint's color data.

Then I added a [`cv::Vec3b rgb = 0;`](https://github.com/AlejandroSilvestri/os1/blob/179e42d8f940530989bae24086e6a339e62f8321/include/MapPoint.h#L380) property to MapPoint class, [passing the color value](https://github.com/AlejandroSilvestri/os1/blob/179e42d8f940530989bae24086e6a339e62f8321/src/LocalMapping.cc#L335) to the constructor.

Finally I [apply color to points in MapViewer](https://github.com/AlejandroSilvestri/os1/blob/179e42d8f940530989bae24086e6a339e62f8321/src/MapDrawer.cc#L80).

Beware in my version `MapDrawer::DrawMapPoints()` is rewritten.