raulmur/ORB_SLAM2 #718: Monocular ORB SLAM fails on KITTI sequence 09

mengweiren, 2019-02-04:
Hi, 

I tested the monocular ORB SLAM on KITTI dataset, but it failed to detect a loop in sequence 09. The predicted result and the ground truth is shown in the figure below (after alignment with the groundtruth using rotation, translation and scaling), which is different from the posted one in your original paper. Can you please explain the reason here? And what can I do to fix the problem? 
[image]
By the way I ran the code ORB SLAM2 without modification but with monocular mode on Ubuntu 18.04.

Appreciate any advice and comments here, thanks!

tyuownu, 2019-02-16:
In the paper, the author said the system not always detect the loop. You can check it out and re-try the dataset.

RyanMiao11, 2020-06-16:
hi,your predicted result  in this figure is  all the pose or only the keyframe? how to save all the pose ? could you give me some advise？ thanks！

stolpa4, 2020-06-16:
@RyanMiao11 To save all the poses for monocular SLAM, you should do two things:
1) use SaveTrajectoryKITTI in the example file instead of SaveKeyFrameTrajectoryTUM
2) Remove these lines:
`if(mSensor==MONOCULAR)
    {
        cerr << "ERROR: SaveTrajectoryKITTI cannot be used for monocular." << endl;
        return;
    }`
from the SaveTrajectoryKITTI function

RyanMiao11, 2020-06-17:
ok，thank you！

Shantnu12345, 2021-11-19:
@stolpa4 I tried doing that but the scale of the results is significantly different than ground truth. Did you face the same issue?

enguang2, 2024-10-27:
@Shantnu12345 This expected if you run with mono camera, where scale is ambiguous, but you can still do the evaluation with a proper scale factor.

supermice, 2024-10-27:
邮件已收到！ ***@***.***

lx-r, 2024-10-27:
您好，来信已收到，谢谢！