raulmur/ORB_SLAM2 #725: Factors that cause scaling error for stereo camera?

YingjieLu, 2019-02-21:
When I tested the code with ZED stereo camera inside building, I found that:
1. In almost all test cases for straight-line movement, the estimated odometry is 5% less than the ground truth. The moving direction is the front of camera. I double checked the baseline setting, and it should not produce that amount of error.
2. When the camera moves besides a wall, the scale is off again. When left camera is closer to the wall, the algorithm tends to under-estimate the result. When right camera is closer to the wall, the algorithm tends to over-estimate the result. "Move besides a wall" means that the camera's facing and moving direction is parallel to the wall.

Anyone has some insights with this problem or met the same issue before? Thanks in advance.

BristolVisualPFT, 2019-11-10:
@YingjieLu I have the same issue as you using the stereo ORB-SLAM. Could you figure out what the issue was? Thanks.

mathuse, 2020-06-08:
I have the stereo camera, and I migrate the code to my code. I find some problem,the first problem is that what time the program calculate the "pKF->mFeatVec" when program first enter the "TrackReferenceKeyFrame/SearchByBoW". I slove it by adding "mCurrentFrame->ComputeBoW();" before "StereoInitialization/new KeyFrame" but it is not perfect, because I have successfully tested stereo camera using the executable file compiled from ORB-SLAM2 code.
The second problem is "std::bad_alloc" when I run the my executing file, and I have find the location of this problem at "TrackReferenceKeyFrame/PoseOptimization/optimizer.optimize(its[it])". I want to know the reason.Can you help me? By the way, I have tried a lot of ways, but none of them worked.