raulmur/ORB_SLAM2 #728: Images turn grayscale when running ORB_SLAM Mono

DigitalGabriele, 2019-02-25:
Hello,

I have recorded some images using ROS. I have run lsd_slam on it and it showed colorful images in the debug viewer. However, when I run ORB_SLAM Mono on the same bag file, it shows me that video as gray. 

Another question. The .yaml files say camera calibration and **distortion** parameters, even though all we adjust are camera intrinsic and projection params (as it notes fx,fy,cx,cy,k1,k2,k3,p1,p2).

hemangchawla, 2019-03-01:
This is by design. In the `Tracking.cc` file, consider the `Tracking::GrabImageMonocular` function. This function converts the image to gray scale by the use of following statements:

```
    mImGray = im;

    if(mImGray.channels()==3)
    {
        if(mbRGB)
            cvtColor(mImGray,mImGray,CV_RGB2GRAY);
        else
            cvtColor(mImGray,mImGray,CV_BGR2GRAY);
    }
    else if(mImGray.channels()==4)
    {
        if(mbRGB)
            cvtColor(mImGray,mImGray,CV_RGBA2GRAY);
        else
            cvtColor(mImGray,mImGray,CV_BGRA2GRAY);
    }
```
This conversion is important for feature extraction. 

However it is have color images in visualization. 
In `Tracking.h`

```
   // Current Frame
    Frame mCurrentFrame;
    cv::Mat mImGray;

   // Add here
    cv::Mat mImOriginal;
```

In `Tracking.cc`, function `Tracking::GrabImageMonocular`

```
    mImGray = im;
    // Add here
    mImOriginal = im;

    if(mImGray.channels()==3)
    {
        if(mbRGB)
            cvtColor(mImGray,mImGray,CV_RGB2GRAY);
        else
            cvtColor(mImGray,mImGray,CV_BGR2GRAY);
    }
    else if(mImGray.channels()==4)
    {
        if(mbRGB)
            cvtColor(mImGray,mImGray,CV_RGBA2GRAY);
        else
            cvtColor(mImGray,mImGray,CV_BGRA2GRAY);
    }

```

In `FrameDrawer.cc` function `FrameDrawer::Update`
```
   unique_lock<mutex> lock(mMutex);
    // pTracker->mImGray.copyTo(mIm); // comment this line
    pTracker->mImOriginal.copyTo(mIm); // add this instead
    mvCurrentKeys=pTracker->mCurrentFrame.mvKeys;

```
Compile again and run. This should show colored image during visualization. 
Note that you may need to modify colors of the map points in FrameDrawer in order to have the contrast between image and the detected features.

DigitalGabriele, 2019-03-01:
Thanks! Do you have any updates about my other question wrt distortion parameters?

hemangchawla, 2019-03-01:
Read this: https://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html

ivannson, 2019-03-01:
@DigitalGabriele Intrinsic parameters are fx, fy, cx, cy. 

k1, k2, k3, p1, p2 are the distortion parameters. 

This provides quite a nice explanation:
https://uk.mathworks.com/help/vision/ug/camera-calibration.html