raulmur/ORB_SLAM2 #732: Failed to load image for KITTI dataset

TangXiaoyu9, 2019-03-01:
Hi, I am having problems with running the example with KITTI dataset, while it works well with EuRoc.

**I have warnings below when compiling:**
txy@txy:~/workspace/ORB_SLAM2$ ./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/txy/workspace/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/txy/workspace/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
CMake Warning at CMakeLists.txt:110 (add_executable):
  Cannot generate a safe runtime search path for target mono_euroc because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libGLU.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/txy/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:85 (add_executable):
  Cannot generate a safe runtime search path for target rgbd_tum because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libGLU.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/txy/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:106 (add_executable):
  Cannot generate a safe runtime search path for target mono_kitti because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libGLU.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/txy/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:91 (add_executable):
  Cannot generate a safe runtime search path for target stereo_kitti because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libGLU.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/txy/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:95 (add_executable):
  Cannot generate a safe runtime search path for target stereo_euroc because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libGLU.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
[... 651 lines cut ...]
7f524beff000-7f524bf46000 r-xp 00000000 08:02 17308710                   /usr/lib/x86_64-linux-gnu/libbluray.so.1.9.2
7f524bf46000-7f524c145000 ---p 00047000 08:02 17308710                   /usr/lib/x86_64-linux-gnu/libbluray.so.1.9.2
7f524c145000-7f524c147000 r--p 00046000 08:02 17308710                   /usr/lib/x86_64-linux-gnu/libbluray.so.1.9.2
7f524c147000-7f524c148000 rw-p 00048000 08:02 17308710                   /usr/lib/x86_64-linux-gnu/libbluray.so.1.9.2
7f524c148000-7f524c192000 r-xp 00000000 08:02 17308713                   /usr/lib/x86_64-linux-gnu/libgme.so.0.6.0
7f524c192000-7f524c392000 ---p 0004a000 08:02 17308713                   /usr/lib/x86_64-linux-gnu/libgme.so.0.6.0
7f524c392000-7f524c395000 r--p 0004a000 08:02 17308713                   /usr/lib/x86_64-linux-gnu/libgme.so.0.6.0
7f524c395000-7f524c396000 rw-p 0004d000 08:02 17308713                   /usr/lib/x86_64-linux-gnu/libgme.so.0.6.0
7f524c396000-7f524c3e0000 r-xp 00000000 08:02 17308729                   /usr/lib/x86_64-linux-gnu/libmodplug.so.1.0.0
7f524c3e0000-7f524c5e0000 ---p 0004a000 08:02 17308729                   /usr/lib/x86_64-linux-gnu/libmodplug.so.1.0.0
7f524c5e0000-7f524c5e1000 r--p 0004a000 08:02 17308729                   /usr/lib/x86_64-linux-gnu/libmodplug.so.1.0.0
7f524c5e1000-7f524c5e2000 rw-p 0004b000 08:02 17308729                   /usr/lib/x86_64-linux-gnu/libmodplug.so.1.0.0
7f524c5e2000-7f524c721000 rw-p 00000000 00:00 0 
7f524c721000-7f524c73c000 r-xp 00000000 08:02 17310489                   /usr/lib/x86_64-linux-gnu/librtmp.so.1
7f524c73c000-7f524c93b000 ---p 0001b000 08:02 17310489                   /usr/lib/x86_64-linux-gnu/librtmp.so.1
7f524c93b000-7f524c93c000 r--p 0001a000 08:02 17310489                   /usr/lib/x86_64-linux-gnu/librtmp.so.1
7f524c93c000-7f524c93d000 rw-p 0001b000 08:02 17310489                   /usr/lib/x86_64-linux-gnu/librtmp.so.1
7f524c93d000-7f524c985000 r-xp 00000000 08:02 17309101                   /usr/lib/x86_64-linux-gnu/libssh-gcrypt.so.4.4.1
7f524c985000-7f524cb84000 ---p 00048000 08:02 17309101                   /usr/lib/x86_64-linux-gnu/libssh-gcrypt.so.4.4.1
7f524cb84000-7f524cb85000 r--p 00047000 08:02 17309101                   /usr/lib/x86_64-linux-gnu/libssh-gcrypt.so.4.4.1
7f524cb85000-7f524cb86000 rw-p 00048000 08:02 17309101                   /usr/lib/x86_64-linux-gnu/libssh-gcrypt.so.4.4.1
7f524cb86000-7f524cba0000 r-xp 00000000 08:02 17306798                   /usr/lib/x86_64-linux-gnu/libcrystalhd.so.3.6
7f524cba0000-7f524cd9f000 ---p 0001a000 08:02 17306798                   /usr/lib/x86_64-linux-gnu/libcrystalhd.so.3.6
7f524cd9f000-7f524cda0000 r--p 00019000 08:02 17306798                   /usr/lib/x86_64-linux-gnu/libcrystalhd.so.3.6
7f524cda0000-7f524cda1000 rw-p 0001a000 08:02 17306798                   /usr/lib/x86_64-linux-gnu/libcrystalhd.so.3.6Aborted (core dumped)

TangXiaoyu9, 2019-03-01:
After several attempts to run KITTI example, I got the information below:

./Examples/Monocular/mono_kitti Vocabulary/ORBvoc.txt Examples/Monocular/KITTI00-02.yaml /home/txy/dataset/KITTI/Odometry/data_odometry_color/dataset/sequences/00

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 718.856
- fy: 718.856
- cx: 607.193
- cy: 185.216
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 10
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 4541

Failed to load image at: /home/txy/dataset/KITTI/Odometry/data_odometry_color/dataset/sequences/00/image_0/000000.png
Segmentation fault (core dumped)

closed 2019-03-05 by TangXiaoyu9

Chrislzy1993, 2019-03-25:
how to you solve it ?? daxiongdi @TangXiaoyu9

tonyeffendi, 2021-03-12:
Rename the folder inside 00:
image_0 -> image_1
image_1 -> image_0

The reason for the crash is because it looks for the image 0000.png but it can't find one

ftzlzw, 2023-04-09:
> 重命名00里面的文件夹： 图片_0 -> 图片_1 图片_1 -> 图片_0
> 
> 崩溃的原因是因为它寻找图像 0000.png 但找不到
just change Image_0 to Image_1 and Image_1 to Image_0?