raulmur/ORB_SLAM2 #741: Monocular SLAM Results on EuRoC dataset

abhilashjain7, 2019-03-15:
Hi!
I cannot find any results for monocular ATE on EuRoC dataset. The ORB-SLAM2 paper gives the results for stereo SLAM on EuRoC.
If someone has also evaluated the code on EuRoC (monocular), could you please validate my results?

| Stream | ATE - RMSE |
|---------|--------------|
| MH_01_easy | 0.037298 |
| MH_02_easy | 0.031539 |
| MH_03_medium | 0.03853 |
| MH_04_difficult | 0.048295 |
| MH_05_difficult | 0.04384 |
| V1_01_easy | 0.088099 |
| V1_02_medium | 0.064248 |
| V1_03_difficult | 0.767821 |
| V2_01_easy | 0.060076 |
| V2_02_medium | 0.0532 |
| V2_03_difficult | 0.164655 |

hoangthien94, 2019-04-12:
This [paper](http://aircconline.com/acij/V9N2/9218acij01.pdf) evaluated ORB-SLAM2 on most popular datasets for (EuRoC, KITTI, TUM). Here are their results on EuRoC: 
[image]

hoangthien94, 2019-04-12:
Have you checked out Raul Mur-Artal's paper ["Visual-Inertial Monocular SLAM with Map Reuse"](https://arxiv.org/pdf/1610.05949.pdf) (see table I, right-most 2 columns)?

For other methods (ROVIO, MSCKF, VINS-Mono etc.), ATE results are readily available in this [paper](https://ieeexplore.ieee.org/document/8460664).

I am curious about how you get the metric scale for monocular ORB-SLAM2 in your results. Did you use the ground truth to scale up, or are you using your own method to calculate true scale from just one camera? If it is the latter, I look forward to seeing your findings.

abhilashjain7, 2019-04-12:
Thanks for linking these two papers. I had somehow missed these two when I was looking for the results.
To get the right scale, I simply used the modified ATE script given [here](https://github.com/raulmur/evaluate_ate_scale), which is a modification of the evaluation script given with the TUM RGB-D dataset. This script automatically aligns the scale of the estimated trajectory to the ground truth for monocular SLAM.

didnotwork, 2020-08-27:
Hello @abhilashjain7, I ran the ORB-SLAM code with EuRoC using both monocular and stereo vision. But I couldn't get overlapping trajectory and ground-truth with any type of camera. Could you draw overlapping trajectories with the ground-truth data? Actually I do not understand to adjust scale factor or anything else manually unless we work with monocular. So, after you create the KeyFrameTrajectory.txt file what kind of a procedure did you follow?