raulmur/ORB_SLAM2 #751: AR demo do not use ROS

jiqirenzhifu, 2019-04-16:
Hello eveyone .I am a newcomer to SLAM, AR demo is very cool, I want to know how to run AR demo without using ROS. such as using a camera or using a video that has already been taken.Any reply is very grateful

EricDJL, 2019-06-28:
作者提供的是ROS版本的AR demo，如果不是用ROS的话，是要自己写的，目前没有尝试过

1628162732, 2020-09-08:
I made a little change to mono_ar.cc
it is works well

#include<iostream>
#include<algorithm>
#include<fstream>
#include<chrono>

#include <ctime>
#include <opencv2/opencv.hpp>
#include<opencv2/core/core.hpp>
#include<opencv2/imgproc/imgproc.hpp>

#include"System.h"

#include"ViewerAR.h"

using namespace std;

ORB_SLAM2::ViewerAR viewerAR;
bool bRGB = true;
const string strVocFile = "../../Vocabulary/ORBvoc.bin";
const string strSettingsFile = "../../Vocabulary/TUM1.yaml";

cv::Mat K;
cv::Mat DistCoef;

class ImageGrabber
{
public:
    ImageGrabber(ORB_SLAM2::System* pSLAM):mpSLAM(pSLAM){}

    //void GrabImage(const sensor_msgs::ImageConstPtr& msg);

    ORB_SLAM2::System* mpSLAM;
};

int main(int argc, char **argv)
{

    // Create SLAM system. It initializes all system threads and gets ready to process frames.
    ORB_SLAM2::System SLAM(strVocFile, strSettingsFile,ORB_SLAM2::System::MONOCULAR,false);

    cout << endl << endl;
    cout << "-----------------------" << endl;
    cout << "Augmented Reality Demo" << endl;
    cout << "1) Translate the camera to initialize SLAM." << endl;
    cout << "2) Look at a planar region and translate the camera." << endl;
    cout << "3) Press Insert Cube to place a virtual cube in the plane. " << endl;
    cout << endl;
    cout << "You can place several cubes in different planes." << endl;
    cout << "-----------------------" << endl;
    cout << endl;

    viewerAR.SetSLAM(&SLAM);

    ImageGrabber igb(&SLAM);

#if 0
	// capture from file
	VideoCapture cap("camera.avi");
#else
	// capture from camera
	cv::VideoCapture cap(1);

	if (!cap.isOpened()) {
		cerr << "Failed to open camera!\n";
		return -1;
	}
#endif

    cv::FileStorage fSettings(strSettingsFile, cv::FileStorage::READ);
    bRGB = static_cast<bool>((int)fSettings["Camera.RGB"]);
    float fps = fSettings["Camera.fps"];
    viewerAR.SetFPS(fps);

    float fx = fSettings["Camera.fx"];
    float fy = fSettings["Camera.fy"];
    float cx = fSettings["Camera.cx"];
    float cy = fSettings["Camera.cy"];

    viewerAR.SetCameraCalibration(fx,fy,cx,cy);

    K = cv::Mat::eye(3,3,CV_32F);
    K.at<float>(0,0) = fx;
    K.at<float>(1,1) = fy;
    K.at<float>(0,2) = cx;
    K.at<float>(1,2) = cy;

    DistCoef = cv::Mat::zeros(4,1,CV_32F);
    DistCoef.at<float>(0) = fSettings["Camera.k1"];
    DistCoef.at<float>(1) = fSettings["Camera.k2"];
    DistCoef.at<float>(2) = fSettings["Camera.p1"];
    DistCoef.at<float>(3) = fSettings["Camera.p2"];
    const float k3 = fSettings["Camera.k3"];
    if(k3!=0)
    {
        DistCoef.resize(5);
        DistCoef.at<float>(4) = k3;
    }

	thread tViewer = thread(&ORB_SLAM2::ViewerAR::Run, &viewerAR);
	bool gFinished = false;
	cv::Mat im;
	while (!gFinished)
	{
		cap >> im;
	
	    cv::Mat imu;
	    cv::Mat Tcw = SLAM.TrackMonocular(im, time(NULL));
	    int state = SLAM.GetTrackingState();
	    vector<ORB_SLAM2::MapPoint*> vMPs = SLAM.GetTrackedMapPoints();
	    vector<cv::KeyPoint> vKeys = SLAM.GetTrackedKeyPointsUn();

	    cv::undistort(im, imu, K, DistCoef);

	    if (bRGB)
		    viewerAR.SetImagePose(imu, Tcw, state, vKeys, vMPs);
	    else
	    {
		cv::cvtColor(imu, imu, CV_RGB2BGR);
		viewerAR.SetImagePose(imu, Tcw, state, vKeys, vMPs);
	    }
	}
	cap.release();
    

    // Stop all threads
    SLAM.Shutdown();

    // Save camera trajectory
    SLAM.SaveKeyFrameTrajectoryTUM("KeyFrameTrajectory.txt");

    return 0;
}