raulmur/ORB_SLAM2 #752: ORB Extractor Parameters - Night

AnneCCollin, 2019-04-16:
Hi,

I'm using ORB-SLAM2 on a sequence that was shot at night, and it seems that ORB is struggling (green lines instead of the squares) and ORBSlam is stuck on "trying to initialize". I can't do much about the data itself, and I am not too familiar with ORB extractors; what is your recommendation to tune the parameters in the .yaml file for a night sequence?

RahmanMifta, 2019-04-17:
Interesting experiment. What type of camera are you using?

AnneCCollin, 2019-06-11:
I'm just running it directly on the nuScenes dataset, one of the night sequences: https://www.nuscenes.org/data-collection

andreabiondo, 2020-07-24:
@AnneCCollin may I ask you what parameters did you use for daylight scenes? I'm also trying to use ORB-SLAM2 in NuScenes but with no luck.