raulmur/ORB_SLAM2 #754: tracking loss

match08, 2019-04-17:
track dataset is "rgbd_dataset_freiburg1_360" loss

daisuke0203, 2019-06-25:
`a`

daisuke0203, 2019-06-25:
`roslaunch chapter5 usb_cam.launch
... logging to /home/daiuske/.ros/log/95689a72-9763-11e9-a31b-080027134c6c/roslaunch-daiuske-VirtualBox-2448.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daiuske-VirtualBox:45640/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /uvc_camera/device: /dev/video0
 * /uvc_camera/format: jpeg
 * /uvc_camera/fps: 30
 * /uvc_camera/height: 480
 * /uvc_camera/width: 640

NODES
  /
    uvc_camera (uvc_camera/uvc_camera_node)

ROS_MASTER_URI=http://localhost:11311

process[uvc_camera-1]: started with pid [2465]
[ INFO] [1561478965.255436526]: using default calibration URL
[ INFO] [1561478965.255787262]: camera calibration URL: file:///home/daiuske/.ros/camera_info/camera.yaml
[ INFO] [1561478965.255966876]: Unable to open camera calibration file [/home/daiuske/.ros/camera_info/camera.yaml]
[ WARN] [1561478965.256078897]: Camera calibration file /home/daiuske/.ros/camera_info/camera.yaml not found.
opening /dev/video0
pixfmt 0 = 'YUYV' desc = 'YUYV 4:2:2'
  discrete: 640x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 160x120:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 176x144:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x176:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/15 1/10 1/5 
  discrete: 960x720:   1/10 1/5 
  discrete: 1024x576:   1/10 1/5 
  discrete: 1184x656:   1/10 1/5 
  discrete: 1280x720:   2/15 1/5 
  discrete: 1280x960:   2/15 1/5 
pixfmt 1 = 'MJPG' desc = 'Motion-JPEG'
  discrete: 640x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 160x120:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 176x144:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x176:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1024x576:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1184x656:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x960:   1/30 1/25 1/20 1/15 1/10 1/5 
  int (Brightness, 0, id = 980900): 0 to 255 (1)
  int (Contrast, 0, id = 980901): 0 to 255 (1)
  int (Saturation, 0, id = 980902): 0 to 255 (1)
  bool (White Balance Temperature, Auto, 0, id = 98090c): 0 to 1 (1)
  int (Gain, 0, id = 980913): 0 to 255 (1)
  menu (Power Line Frequency, 0, id = 980918): 0 to 2 (1)
    0: Disabled
    1: 50 Hz
    2: 60 Hz
  int (White Balance Temperature, 16, id = 98091a): 0 to 10000 (10)
  int (Sharpness, 0, id = 98091b): 0 to 255 (1)
  int (Backlight Compensation, 0, id = 98091c): 0 to 1 (1)
  menu (Exposure, Auto, 0, id = 9a0901): 0 to 3 (1)
  int (Exposure (Absolute), 16, id = 9a0902): 1 to 10000 (1)
  bool (Exposure, Auto Priority, 0, id = 9a0903): 0 to 1 (1)
  menu (LED1 Mode, 0, id = a046d05): 0 to 3 (1)
    0: Off
    1: On
    2: Blink
    3: Auto
  int (LED1 Frequency, 0, id = a046d06): 0 to 131 (1)
terminate called after throwing an instance of 'std::runtime_error'
  what():  unable to start capture
[uvc_camera-1] process has died [pid 2465, exit code -6, cmd /opt/ros/kinetic/lib/uvc_camera/uvc_camera_node image_raw:=/uvc_camera/image_raw image_raw/compressed:=/uvc_camera/image_raw/compressed camera_info:=/uvc_camera/camera_info __name:=uvc_camera __log:=/home/daiuske/.ros/log/95689a72-9763-11e9-a31b-080027134c6c/uvc_camera-1.log].
log file: /home/daiuske/.ros/log/95689a72-9763-11e9-a31b-080027134c6c/uvc_camera-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

daisuke0203, 2019-06-25:
`ggggjvvj j
k bk`