raulmur/ORB_SLAM2 #755: Installation error: OpenCV version

petroskey, 2019-04-24:
Hi everyone, thank you always for your help.

The problem comes out when I compile this great software on Ubuntu18.04.
The error is as follows:
-- Configuring done
-- Generating done
-- Build files have been written to: 
/home/user-name/orb_slam2_ws/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
CMake Error at CMakeLists.txt:35 (message):
  OpenCV > 2.4.3 not found.

-- Configuring incomplete, errors occurred!

The OpenCV version is 4.1.0-dev.

Thank you for advance.

DKandrew, 2019-05-20:
Hi @petroskey I think ORB SLAM2 doesn't not support OpenCV 4.1 for the time being. 
One big problem is that PnPsolver does not support OpenCV 4.1, many OpenCV APsI it uses are removed in the OpenCV 4.   

You can try it out yourself by changeing the version of OpenCV to 4.0 in the line 31 of CMakeLists.txt.

Windfisch, 2019-07-17:
I think I fixed this in #790 or https://github.com/Windfisch/ORB_SLAM2
At least it works for me under Arch Linux with OpenCV 4.1.0.