raulmur/ORB_SLAM2 #758: Include intrinsics in Bundle Adjustment

alessandroTorresani, 2019-04-29:
I was wondering if it is possible to optimise also the intrinsic parameters of the camera (fx, fy, cx, cy) during the Bundle Adjustment (local and global). Since ORB-SLAM is using g2o to solve BA, I was checking the g2o API but i didn't find any reference to intrinsic optimisation. Has someone of you already managed to solve that?

AlejandroSilvestri, 2020-02-14:
@alessandroTorresani 

It may be possible, but is a different optimisation.  You should rewrite the code, specially when defining de optimization problem en bundle adjustment.

In my experience distortion parameters are very important.  If you manage to get intrinsics automatically, remember you already must calibrate the camera for the distortion parameters.

alessandroTorresani, 2020-03-02:
Thank you @AlejandroSilvestri. I suspected  indeed that it wasn't easy.