raulmur/ORB_SLAM2 #759: roslib.manifest wrong

syp1997, 2019-04-30:
When I Execute build_ros.sh, the error like this. How can I solve this problem? Thanks very much!

[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from syntax check of ORB_SLAM2/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named roslib.manifest
CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/private.cmake:78 (message):
  [rosbuild] Syntax check of ORB_SLAM2/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!

Vct0ry, 2019-12-08:
hellow, have your problem fixed? I meet the same trouble, could you give me some advise?

disini, 2020-04-07:
hello buddy, I meet the same problem too, have your fixed it yet?

EmreOzkose, 2020-06-15:
Hello everyone, I have the same problem too. Is there a solution?

karnikram, 2020-08-24:
I had the same problem.  Somehow `roslib` couldn't be found even after all the environment variables were set properly. (`rospack find roslib`). This is a core ROS package so as a lazy solution I just reinstalled ROS and everything started to build properly.

saleelirenyun, 2021-04-29:
Hi, have you resolve this error?

JsutCheng, 2021-12-13:
I fond that I can import roslib in the terminal, so I find the python interpreter. Then you need to go to this file /opt/ros/noetic/share/ros/core/rosbuild/private.cmake, about line 67 COMMAND ${PYTHON_EXECUTABLE} -c "import roslib.manifest; roslib.manifest.parse_file('manifest.xml')", just change ${PYTHON_EXECUTABLE}  with /usr/bin/python3.8, then it may work.

Vct0ry, 2021-12-13:
收到，谢谢！——自动回复。

fennushapaozi, 2022-03-10:
我在编译DS_SLAM时，也遇到了同样的问题。重装了ROS，没有用。按照楼上修改了private.cmake，也不行。系统为Ubuntu20.04。

Vct0ry, 2022-03-10:
收到，谢谢！——自动回复。

Taomin00, 2022-03-17:
> private.cmake

请问您解决这个问题了吗

fennushapaozi, 2022-03-17:
它的问题是python要导入的包找不到，这个包是在roslib文件夹下的manifest.py文件。所以只需要在.bashrc文件中添加环境变量PYTHONPATH="${PYTHONPATH}:/opt/ros/noetic/lib/python3/dist-packages"
export PYTHONPATH
即可（我的ROS版本是noetic）。

GuoJeffrey, 2022-09-19:
> 它的问题是python要导入的包找不到，这个包是在roslib文件夹下的manifest.py文件。所以只需要在.bashrc文件中添加环境变量PYTHONPATH="${PYTHONPATH}:/opt/ros/noetic/lib/python3/dist-packages" export PYTHONPATH 即可（我的ROS版本是noetic）。

需要按安装位置改动dist_packages路径

Vct0ry, 2022-09-19:
收到，谢谢！——自动回复。

Vct0ry, 2022-10-11:
收到，谢谢！——自动回复。