raulmur/ORB_SLAM2 #762: ORS-SLAM2 breaks off with error:double free or corruption (!prev)

TianlunHu, 2019-05-15:
Hi guys, 

I'm a fresh newbie to SLAM and I'm trying to run this ORB_SLAM2 with the Sequences of my own.  The system could be started correctly, but it always flash out when it trying to initialize the map. And come up with the following errors:
++++++++++++++++++++++++++++++++++++++++++++++++++++++++
-------
Start processing sequence ...
Images in the sequence: 5199

*** Error in `./Examples/Monocular/mono_tum': double free or corruption (!prev): 0x000000000f3f8720 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f91b93557e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f91b935e37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f91b936253c]
/home/tianlunhu/Documents/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so(_ZN3g2o10HyperGraph5clearEv+0x79)[0x7f91b8e8a9a9]
......
======= Memory map: ========
00400000-0040e000 r-xp 00000000 08:09 526454                             /home/tianlunhu/Documents/ORB_SLAM2/Examples/Monocular/mono_tum
0060d000-0060e000 r--p 0000d000 08:09 526454                             /home/tianlunhu/Documents/ORB_SLAM2/Examples/Monocular/mono_tum
0060e000-0060f000 rw-p 0000e000 08:09 526454                             /home/tianlunhu/Documents/ORB_SLAM2/Examples/Monocular/mono_tum
00ba4000-0f983000 rw-p 00000000 00:00 0                                  [heap]
7f9144000000-7f9144025000 rw-p 00000000 00:00 0 
7f9144025000-7f9148000000 ---p 00000000 00:00 0 
7f9148000000-7f9148025000 rw-p 00000000 00:00 0 
7f9148025000-7f914c000000 ---p 0000'0000 00:00 0 
7f914c000000-7f914c025000 rw-p 00000000 00:00 0 
7f914c025000-7f9150000000 ---p 00000000 00:00 0 
7f9151186000-7f91512dd000 rw-s 00000000 00:05 5177349                    /SYSV00000000 (deleted)
.........
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
I've tried it for Mono, Streo and ROS modules, but all failed. I thought there might be something wrong with the lib files in my system. 
Did anyone have some idea about this ?

Thanks

TianlunHu, 2019-05-16:
Problem solved, thanks

den250400, 2019-06-02:
Can you provide a brief review on how you solved this problem?

linhan94, 2019-10-15:
@TianlunHu How did you solve it?

grafoteka, 2019-12-10:
Hello, does anyone how to solve this problem?

1023198294, 2020-03-13:
I have the same problem too, did anyone get through the same problem?

PushyamiKaveti, 2020-09-09:
I have the same problem suddenly, it was working well before. How did you solve it?

PushyamiKaveti, 2020-09-09:
SOLVED!!
The error for me was from g2o library. Once I compiled it with cmake option -march=native it worked for me.

AQIMO, 2023-09-01:
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY "${PROJECT_SOURCE_DIR}/Examples/Stereo -march=native")