raulmur/ORB_SLAM2 #768: Failed to load image, how to slove!!

weiconh, 2019-05-26:
When I run Monocular demo，everything is ok. however, when I try to run the Stereo demo，an error happened.It failed to load the image. The stereo dataset I downloaded from EuRoC Dataset and I run the demo on VM ubuntu16.04, 
`wei@wei-virtual-machine:~/ORBSLAM/ORB_SLAM2$ ./Examples/Stereo/stereo_euroc Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml /home/wei/ORBSLAM/ORB_SLAM2/senors/cam0/data /home/wei/ORBSLAM/ORB_SLAM2/senors/cam1/data Examples/Stereo/EuRoC_TimeStamps/SEQUENCE.txt

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.85272

-------
Start processing sequence ...
Images in the sequence: 3682

Failed to load image at: /home/wei/ORBSLAM/ORB_SLAM2/senors/cam0/data/1403636579763555584.000000 1403636579763555584.png 1403636579763555584.000000 1403636579763555584.png.png
`

weiconh, 2019-05-26:
I slove！

closed 2019-05-26 by weiconh

deepakyeleshetty, 2019-06-18:
@weiconh Could you please explain how?