raulmur/ORB_SLAM2 #772: Not processing all images in RGB mode.

immuhammadadil, 2019-06-18:


Alexandre34, 2019-06-28:
Hello @immuhammadadil 
In your configuration file (file with extension yaml), have you checked if the **Camera.fps** parameter is the same value as the framerate of your camera? If it is lower then the system might skip frames.
As for the trajectory file, I have the same behaviour as yours. The system only keeps frames that are relevant for the trajectory (for instance, with 1800 frames, only 50 are kept).

closed 2019-06-29 by immuhammadadil

Alexandre34, 2019-07-01:
@immuhammadadil 
Can you please post how you have solved the problem? Other people might have the same problem as you.

vivekgoel2212, 2019-07-05:
@Alexandre34 
Can you please elaborate how **Camera.fps** can be problematic?
Suppose I capture the video **@20fps** and using **Camera.fps = 10** then in which cases the system will skip the frames?

Alexandre34, 2019-07-05:
@vivekgoel2212 
From my experience, having **Camera.fps** lower that the actual frame rate of the camera means that the system will skip some frames of your recording. In your example, it will only take half of your frames to the system, the other half won't be used.
Try to change the parameter to a very low value (like 3 fps) and launch the program. The viewer will be extremely slow and it will have trouble initializing itself.

vivekgoel2212, 2019-07-05:
@Alexandre34 
I didn't experience this behavior. I think to study this problem we should first differentiate between two cases.
1. Real time camera footage with default camera fps.
2. Video capture and saved with the same camera with different fps from the default.
Also we should discuss the significance of [this ](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Tracking.cc#L85)comment taking above points in focus.