raulmur/ORB_SLAM2 #774: Failed to load Image

deepakyeleshetty, 2019-06-21:
NOTE: Issue is the same as https://github.com/raulmur/ORB_SLAM2/issues/768, but it was closed without a solution. Was wondering if there is any solution.

I tried to run ORB_SLAM2 stereo example for the EuRoC dataset V2_01. The terminal throws an error "Failed to load image at \path_to_the_first_image_in_the_sequence\"

The message is similar to the one referenced above. Could anyone please let me know where I made a mistake?

Thanks!

gsenkowski, 2019-06-25:
Hello,

maybe there is an additional '/' somwhere in the path to the images wich leads to the double '/' after 'data' before '1413393212255760384.png' in
/dataset/mav0/cam0/data//1413393212255760384.png

deepakyeleshetty, 2019-06-28:
Hello @gsenkowski ,

Thanks a lot for pointing it out! I completely overlooked it. It now works and I am closing the issue.

closed 2019-06-28 by deepakyeleshetty