raulmur/ORB_SLAM2 #777: Not able to catch the feature in the all the frames

ajay1606, 2019-06-29:
Hello everyone, 
Thanks for the nice ORB_SLAM2 opensource package. I have tested the running the ORB_SLAM2 with ROS using the sample dataset provided in the https://github.com/raulmur/ORB_SLAM2.

With the sample dataset, it's working well, but with my own data is partially working.  With my dataset, unable to catch the feature in some image sequences. In the images, it's showing straight green lines in some image sequence. As shown in the figure below. 
**Failed Case:**
[image]

**Successful case:**
But it's working well with the few image sequence as shown below.
[image]

Could anyone please suggest me, cause of this issue? Is it because of vocabulary? Currently, I am using vocabulary provided with ORB_SLAM2 package. Is there any tutorial which gives detailed information about train our own vocabulary? 

Is this issue not because of the vocabulary could anyone please recommend a possible fix. 
Your help much appreciated.

WOrking Platform:
Ubuntu (16.04), ROS (Kinetic), ORB_SLAM2 (ROS), MONO, Input Method (ros bag), FPS (30)

Thanks
Ajay

Alexandre34, 2019-07-01:
Hello @ajay1606 
The green lines that you have on your dataset means that the system is initializing but can't find the landmarks (green squares).
I have the same issue with some of my datasets. In the beginning I try to let the system initialize by not moving the camera in the few first seconds. A quick fix that is working alright is to raise the **nFeatures** parameter in your config file.
```
# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 3000
```
The higher it is, the more the system will find landmarks, but this will raise the load on your CPU.

I also saw that low contrast frames can also cause issues.

Let me know if this helps.

ajay1606, 2019-07-01:
@Alexandre34 Thank you so much for your response. You said absolutely right, feature capturing working well with parameter ORBextractor.nFeatures: 3000. Thank you so much.

By the way, do you have any idea how to create vocabulary .txt file ? is there any library or tool available?

Because I would like to create Vocabulary on my own using the local feature. That may work well with even  ORBextractor.nFeatures: 2000 (Optimal number features).

Also, I will consider your comment on the contrast and I will test in high contrast.

Thanks a lot for your suggestion.

Alexandre34, 2019-07-01:
@ajay1606 
For the vocabulary file, I have no idea how to customize it. Personally I use the included one and it fits my needs.

Great that it worked! Be aware that this method is very CPU intensive, 3000 is the maximum that worked on my system, your mileage may vary.

From my experience, high frame rate and high contrast gives the best results.

ajay1606, 2019-07-02:
@Alexandre34 Thank you so much for your valuable inputs. I will consider your suggestions.

Thank you so much again :)