raulmur/ORB_SLAM2 #778: Build failed Ubuntu 18.04

angussmitchell, 2019-06-30:
Build fails on Ubuntu18.04.
I followed instructions to install all dependencies. I didn't build any dependencies from source - I went with a apt-get versions for example "apt-get install libopencvdev" "apt-get install libeigen3-dev"

I would appreciate any feedback, I find the following output quite un-descriptive.

After running ./build.sh I get:

`Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/a/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/a/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/a/ORB_SLAM2/build
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
In file included from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/a/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/a/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/a/ORB_SLAM2/include/Tracking.h:31,
                 from /home/a/ORB_SLAM2/include/System.h:29,
                 from /home/a/ORB_SLAM2/src/System.cc:23:
/home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
                                                                          ^~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:344:0,
                 from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
[... 330 lines cut ...]
/home/a/ORB_SLAM2/src/System.cc:236:17: note: suggested alternative: ‘fseek’
                 usleep(1000);
                 ^~~~~~
                 fseek
/home/a/ORB_SLAM2/src/System.cc: In member function ‘void ORB_SLAM2::System::Shutdown()’:
/home/a/ORB_SLAM2/src/System.cc:309:13: error: ‘usleep’ was not declared in this scope
             usleep(5000);
             ^~~~~~
/home/a/ORB_SLAM2/src/System.cc:309:13: note: suggested alternative: ‘fseek’
             usleep(5000);
             ^~~~~~
             fseek
/home/a/ORB_SLAM2/src/System.cc:315:9: error: ‘usleep’ was not declared in this scope
         usleep(5000);
         ^~~~~~
/home/a/ORB_SLAM2/src/System.cc:315:9: note: suggested alternative: ‘fseek’
         usleep(5000);
         ^~~~~~
         fseek
CMakeFiles/ORB_SLAM2.dir/build.make:62: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/System.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2`

giwrgosg, 2019-07-02:
Insert the following:

> #include <unistd.h>

in system.h

angussmitchell, 2019-07-12:
Thanks @giwrgosg , the following worked for me on Ubuntu18.04

closed 2019-07-12 by angussmitchell

whu-lyh, 2019-07-27:
Thanks for this solution, but by the way, i wander why this error occurs on Ubuntu18.04 while i didn't meet this error compiling on ubuntu16.04～～

Dennis2Choi, 2019-12-22:
Thanks a lot 
solution ( add #include <unistd.h> in system.h ) works well on Ubuntu 18.04

Songtingt, 2020-06-19:
Thanks @giwrgosg ！！！！！！

75kgSLAM, 2022-02-20:
> Insert the following:
> 
> > #include <unistd.h>
> 
> in system.h

Thanks! But I wander where to find system.h?

lx-r, 2022-02-20:
您好，来信已收到，谢谢！

aseslamian, 2022-03-07:
> > Insert the following:
> > > #include <unistd.h>
> > 
> > 
> > in system.h
> 
> Thanks! But I wander where to find system.h?

You can easily search in ORB_SLAM 2 folder at top ..... it is located in "include" folder.

Fei-Work, 2024-04-03:
> Insert the following:
> 
> > #include <unistd.h>
> 
> in system.h

It works. But anybody know why it will happen？