raulmur/ORB_SLAM2 #782: fatal error: opencv/cv.h: No such file or directory  #include <opencv/cv.h>

Tengyun-Mo, 2019-07-04:
I have installed opencv4.1.1 on my PC,the opencv4.1.1 don't include the header file named cv.h.so which header file should I replace opencv/cv.h?

Tengyun-Mo, 2019-07-04:
`/home/ubuntu/Downloads/ORB_SLAM2/include/ORBextractor.h:26:10: fatal error: opencv/cv.h: No such file or directory
 #include <opencv/cv.h>
          ^~~~~~~~~~~~~
compilation terminated.`

Windfisch, 2019-07-17:
I think I fixed this in #790 or https://github.com/Windfisch/ORB_SLAM2
At least it works for me under Arch Linux with OpenCV 4.1.0.

ckchng, 2019-11-05:
Hi @Windfisch Thank you so much for your effort. By the way I'm having problem in finding 'opencv2/imgcodecs/legacy/constancts_c.h' while compiling it with OpenCV 4.0.0. Do you have any idea on top of your head why this might be a problem?

ckchng, 2019-11-05:
I managed to 'solve' it by 
1) removing that inclusion (#include "opencv2/imgcodecs/legacy/constancts_c.h").
2) and https://github.com/raulmur/ORB_SLAM2/issues/451#issue-268699827

Thanks.

Windfisch, 2019-11-06:
Hi @ckchng, it appears to me that the constants [have been moved](https://github.com/opencv/opencv/commit/a574788e89af618effd5067063da797180f438c6#diff-620e01e5200e71af4662d455f6a615a3) from `modules/imgcodecs/include/opencv2/imgcodecs.hpp` to `modules/imgcodecs/include/opencv2/imgcodecs/legacy/constants_c.h` on 2018-11-17, and opencv 4.0.0 has been released one day later. So it should™ work for you.
Are you sure that you're using OpenCV 4.0.0, and not a release-candidate (e.g. 4.0.0-rc1)?
Which OS/Distro in which version do you use, and where did you get OpenCV from? It it possible that cmake decided not to use your OpenCV4 installation, but an old  OpenCV3 installation that might be still flying around somewhere?