raulmur/ORB_SLAM2 #784: No output when running ORB-SLAM2 with rgbd_dataset_freiburg1_xyz

HesperX, 2019-07-08:
I try to run ORB-SLAM2 with dataset rgbd_dataset_freiburg1_xyz but nothing happens after I enter the instruction. 
I am with SLAM so I don’t know why it comes out and how to fix it either.
Thank you very much.
[image]

Alexandre34, 2019-07-08:
Hello @HesperX 
You need to change **TUMX** to either **TUM1**, **TUM2** or **TUM3** depending on the dataset. In your case that should be the first one.
There is also the association file that is missing from the command.

HesperX, 2019-07-08:
> Hello @HesperX
> You need to change TUMX to either TUM1, TUM2 or TUM3 depending on the dataset. In your case that should be the first one.
> There is also the association file that is missing from the command.

Oh, I missed it. It works now. 
Thank you very much for the timely help!

closed 2019-07-08 by HesperX

xiaodongww, 2021-09-07:
@yichen928  Hi, do you know where to download the `association file`? I am new in this area and can not find it on TUM website. Could you give me some advice?

zhouwuxiong, 2022-07-07:
I recently had this problem. The reason I had this is that I forgot to decompress "rgbd_dataset_freiburg1_xyz". Because it doesn't have a .gz extension, it's easy to mistake it for a special file that is automatically parsed by the program, but in fact it requires us to do a "tar" command to compress it. (Note that the decompressed folder will replace the original file, since they have the same name). Then execute the mono_tum program in the Example file and everything works fine for me.
If you also encounter this problem, I hope my encounter can help you.