raulmur/ORB_SLAM2 #787: ORB SLAM2 mono no response

eeyzl5, 2019-07-16:
When I tried to run monocular orb slam2 on both TUM and KITTI datasets, there is no any response, no error, no display and just stuck without any output. I also tried the ROS version and it has the same problem. However, when I then tried stereo orb slam2 with EuRoC dataset and everything just went well. So I am not sure if it's the problem of the dataset or the installation of orb slam2 mono. Very glad if anyone could give me some hints.

Alexandre34, 2019-07-16:
Hello @eeyzl5 
Can you give the command that you run? Which dataset in TUM are you running?
Thanks

eeyzl5, 2019-07-16:
Hi @Alexandre34 

For the mono orb slam2 I just ran the command provided in README:
`./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml home/daniela/Downloads/rgbd_dataset_freiburg1_xyz/rgb`

I was using freiburg1 dataset.

For the ROS version I ran a node like this:
`rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml`

Then play the rosbag:
`rosbag play rgbd_dataset_freiburg1_desk.bag`

Both attempts failed without any response and output.

Thanks!

Alexandre34, 2019-07-17:
@eeyzl5 
Try `./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml home/daniela/Downloads/rgbd_dataset_freiburg1_xyz` (without the /rgb at the end)
By default the monocular mode reads the file **rgb.txt** in the folder sequence specified then reads the images. Because there are no **rgb.txt** file in the rgb folder it doesn't read anything.

eeyzl5, 2019-07-17:
@Alexandre34 

Thanks for your advice but unfortunately it's still the same problem. Probably it's the problem of the installation?

Alexandre34, 2019-07-18:
@eeyzl5 
Ok that's weird. Try to reinstall the library and make sure that OpenCV and Eigen are of the right versions.

eeyzl5, 2019-07-19:
@Alexandre34 

Oh! I eventually find out the stupid mistake. I missed a '/' before 'home' that specifies the path to sequence folder. After I add the '/' everything just works fine. Thank you so much for your help!