raulmur/ORB_SLAM2 #793: error in make step

zcoo, 2019-07-23:
I have installed prerequisites. However when I run "./build.sh", it failed. I went into "build.sh" and run it line by line and found it unsuccessful at last line, namely "make" step after "cmake" of ORB_SLAM2  [line 31](https://github.com/raulmur/ORB_SLAM2/blob/master/build.sh#line31)
I search for the solution in Issues but fail to solve. 
Environment:
Ubuntu 16.04 in Virtual Machine( vmware workstation)
opencv3.2.0
gcc version 5.4.0 20160609  (I guess if it is a problem about gcc version?)

```
zc@ubuntu:~/ORB_SLAM2/build$ sudo make
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
	c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 4
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
zc@ubuntu:~/ORB_SLAM2/build$ 

```

I find no useful output above.  Thanks for any help!

Alexandre34, 2019-07-24:
Hello @zcoo 
To me this looks like a RAM issue. The compiler gcc tends to be quite thirsty when it comes to memory, how much memory have you allocated to the VM?

zcoo, 2019-07-24:
@Alexandre34 
Oh, thank you for your help!!! It is really an out of memory problem! I reallocate 3G rather than 1G and make successfully! Thank you very much!

closed 2019-07-24 by zcoo

Esirn, 2020-11-21:
@Alexandre34 
☆⌒(*＾-゜)v THX!! I gave it 8GB, but with the Firefox opening hahahahaha