raulmur/ORB_SLAM2 #794: Run stereo ORBSLAM2，why is the coordinates so small?

HKVINS89757, 2019-07-24:
I run stereo orbslam2  
And compare it with cartographer 
What I can confirm is that cartographer can be used as an accurate standard.

The trajectory of two methods above as follows：
[image]

[image]

thohemp, 2019-07-31:
Have you adapt the specific calibration information for your camera?

AnfangRobkit, 2019-10-23:
Sorry for my dummy questions:
But how could you get the trajectory? Any hint is wellcome!
Did you reconstruct it from KeyFrameTrajectory.txt ?

thohemp, 2019-10-23:
@AnfangRobkit Yes, check this: https://github.com/MichaelGrupp/evo

AnfangRobkit, 2019-10-23:
> @AnfangRobkit Yes, check this: https://github.com/MichaelGrupp/evo

Great hints! And thanks for your reply!

One more question, the KeyFrame has obvious fewer points than the ground truth, but due to the x-axis is timestamp, so it did not affect the result? Am I right?

Anyway, still great thx for your information.!

mathuse, 2020-06-08:
> @AnfangRobkit Yes, check this: https://github.com/MichaelGrupp/evo

I have the stereo camera, and I migrate the code to my code. I find some problem,the first problem is that what time the program calculate the "pKF->mFeatVec" when program first enter the "TrackReferenceKeyFrame/SearchByBoW". I slove it by adding "mCurrentFrame->ComputeBoW();" before "StereoInitialization/new KeyFrame" but it is not perfect, because I have successfully tested stereo camera using the executable file compiled from ORB-SLAM2 code.
The second problem is "std::bad_alloc" when I run the my executing file, and I have find the location of this problem at "TrackReferenceKeyFrame/PoseOptimization/optimizer.optimize(its[it])". I want to know the reason.Can you help me? By the way, I have tried a lot of ways, but none of them worked.