raulmur/ORB_SLAM2 #800: OpenCV4 support

chase-metzger, 2019-08-18:
I would just like to confirm this library and examples were able to build (lots of headaches) and run the examples on Fedora 30 with OpenCV4 (had to build myself with Qt as the gui). If this is relevant to anyone or would like to know more about my setup I can provide more info as needed.

# Screenshot of running mono_euroc example

[image]

samirbr, 2019-10-28:
@chase-metzger can you share your setup, patches, or code changes? Thanks!

tooHotSpot, 2020-01-23:
@chase-metzger Can you share your setup, please? Main headache is in configuring proper CMakeLists and redirecting to folder usr/local/include/opencv4/opencv2 where OpenCV 4.0.1 is installed.

thohemp, 2020-02-07:
Check this: https://github.com/raulmur/ORB_SLAM2/pull/790/

Mechazo11, 2022-10-11:
> Check this: #790

@thohemp the status on this commit still shows Open https://github.com/raulmur/ORB_SLAM2/pull/790/files. If we manually replicate these changes in appropriate files on top of a clean copy of ORB-SLAM2 would that make OpenCV 4.0 work?