raulmur/ORB_SLAM2 #803: initialization is slow

IvyGongoogle, 2019-08-23:
hello, when I use `rgbd_dataset_freiburg1_xyz` datasets of TUM for testing, I find the initialization is slow, about several seconds. Is this normal?

Alexandre34, 2019-08-23:
Hi @IvyGongoogle 
An initialization of a few seconds is normal. My personal experience with it is that most of the time, it takes between 5 to 10 seconds to initialize.
It also depends on if the camera is moving or not and the complexity of the scene. If you place the camera in front of a white wall, it will have trouble initializing.

IvyGongoogle, 2019-08-26:
@Alexandre34  Thanks for your reply. I find when it initializes successfully the camera move a lot `translation`, which is not acceptable. Having a little translation when initializing successfully is expected.

AlejandroSilvestri, 2020-02-14:
@IvyGongoogle 

You need a big deal of translation to initialize.  Actually, you need parallax to triangulate the first mappoints.  This mean you need big "relative" translation.  You can accomplish this with little translation in a nearby scene.

mathuse, 2020-06-08:
> @Alexandre34 Thanks for your reply. I find when it initializes successfully the camera move a lot `translation`, which is not acceptable. Having a little translation when initializing successfully is expected.

I have the stereo camera, and I migrate the code to my code. I find some problem,the first problem is that what time the program calculate the "pKF->mFeatVec" when program first enter the "TrackReferenceKeyFrame/SearchByBoW". I slove it by adding "mCurrentFrame->ComputeBoW();" before "StereoInitialization/new KeyFrame" but it is not perfect, because I have successfully tested stereo camera using the executable file compiled from ORB-SLAM2 code.
The second problem is "std::bad_alloc" when I run the my executing file, and I have find the location of this problem at "TrackReferenceKeyFrame/PoseOptimization/optimizer.optimize(its[it])". I want to know the reason.Can you help me? By the way, I have tried a lot of ways, but none of them worked.