raulmur/ORB_SLAM2 #807: reading camera pose

igallagi23, 2019-09-01:
hi, i know how to get the camera pose(as in this video https://www.youtube.com/watch?v=KkRIkCCW79A) and use the RV . but i can't really understand it.
i want to use it and be able to read it as 3D(XYZ) format .
is there a function or an explanation for this?
it is ok if i can use it in the regular getPose format but i would love for an explanation

AlejandroSilvestri, 2020-02-14:
@igallagi23 

tracker->mCurrentFrame.getCameraCenter() will give you the position vector.

3D poses are 4x4 matrices in homogeneous coordinates.  Last row is 0,0,0,1.  Top 3 rows are Rotation matrix and a 4º column with position (translation).

It gets harder and harder.  I warn you, it's like a Pandora's box.

shiniankanchaige, 2020-05-18:
> hi, i know how to get the camera pose(as in this video https://www.youtube.com/watch?v=KkRIkCCW79A) and use the RV . but i can't really understand it.
> i want to use it and be able to read it as [...]

hello i   met the same  problem ，have you sloved it？