raulmur/ORB_SLAM2 #810: Segmentation Fault (Core Dumped) with Stereo KITTIXX

YumaTheCompanion, 2019-09-04:
I have a working ORB_SLAM2 installation on Ubuntu 18.04. Now, working means, it works under vanilla build. I am working on a custom recorded stereo grayscale image set, that is prepared as KITTI dataset requires.

Now, I wanted to modify this piece of code in the file `Examples/Stereo/stereo_kitti.cc`

```cpp
        // Wait to load the next frame
        double T=0;
        if(ni<nImages-1)
            T = vTimestamps[ni+1]-tframe;
        else if(ni>0)
            T = tframe-vTimestamps[ni-1];

        if(ttrack<T)
            usleep((T-ttrack)*1e6);
```

Because currently, `usleep((T-ttrack)*1e6);` is just too much time. The frames are at 24 FPS, so, 42.6 ms should be enough wait between them, instead, I get 1-2 seconds of wait between frames, in the vanilla setup. 

(My PC is intel i7, 16Gb RAM nvidia GTX 1050 Mobile, the frames are 650x320, nPoints=6000
Is 1-2 frame per second normal for this? I think it's quite a lot!)

Then, I went ahead and did something like this:
```cpp
usleep(43*1e3); // This makes it wait 43 ms per frame.
```

However, to my surprise, I'm getting a segfault (core dumped) error at frame 149. I compiled everything again with debugging symbols and ran with `gdb`. Here is the backtrace (`bt` command):

```
(gdb) bt
#0  0x00007ffff5813fa0 in __GI___pthread_mutex_lock (mutex=0x640) at ../nptl/pthread_mutex_lock.c:65
#1  0x00007ffff790a8c6 in __gthread_mutex_lock (__mutex=0x640) at /usr/include/x86_64-linux-gnu/c++/7/bits/gthr-default.h:748
#2  0x00007ffff790a8c6 in std::mutex::lock() (this=0x640) at /usr/include/c++/7/bits/std_mutex.h:103
#3  0x00007ffff790a8c6 in std::unique_lock<std::mutex>::lock() (this=0x7ffffffd8300) at /usr/include/c++/7/bits/std_mutex.h:267
#4  0x00007ffff790a8c6 in std::unique_lock<std::mutex>::unique_lock(std::mutex&) (__m=..., this=0x7ffffffd8300)
    at /usr/include/c++/7/bits/std_mutex.h:197
#5  0x00007ffff790a8c6 in ORB_SLAM2::KeyFrame::GetPose() (this=this@entry=0x0) at /mnt/hdd/ORB_SLAM2/src/KeyFrame.cc:88
#6  0x00007ffff78bf863 in ORB_SLAM2::Tracking::UpdateLastFrame() (this=this@entry=0x55555afbeac0)
    at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:820
#7  0x00007ffff78c01c7 in ORB_SLAM2::Tracking::TrackWithMotionModel() (this=this@entry=0x55555afbeac0)
    at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:886
#8  0x00007ffff78c1b40 in ORB_SLAM2::Tracking::Track() (this=this@entry=0x55555afbeac0) at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:322
#9  0x00007ffff78c3241 in ORB_SLAM2::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&) (this=0x55555afbeac0, imRectLeft=..., imRectRight=..., timestamp=@0x7fffffffd7f8: 310.15600000000001) at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:205
#10 0x00007ffff78abf08 in ORB_SLAM2::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&) (this=0x7fffffffd990, imLeft=..., imRight=..., timestamp=@0x7fffffffd7f8: 310.15600000000001) at /mnt/hdd/ORB_SLAM2/src/System.cc:174
#11 0x0000555555557a79 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /mnt/hdd/ORB_SLAM2/Examples/Stereo/stereo_kitti.cc:100
```

And the sigsev code, pointed out by `gdb` is this:

```cpp
cv::Mat KeyFrame::GetPose()
{
    unique_lock<mutex> lock(mMutexPose);  // <== This is where sigsev occurs
    return Tcw.clone();
}
```

Now, I fiddled around with the code a bit and changed it to this:
```cpp
cv::Mat KeyFrame::GetPose()
{
    unique_lock<mutex> lock1(mMutexPose, std::defer_lock);

    std::cout << "Arrived at GetPose()." << std::endl;

    if(!Tcw.empty()) {
        std::cout << "Tcw is NOT empty!" << std::endl;
        return Tcw.clone();
    }
    else {
        std::cout << "Tcw is empty!" << std::endl;
        return Tcw.clone();
    }
}
```

This time, the code gets **sigsegv** at frame 149, again, but with this output
```
Thread 1 "stereo_kitti" received signal SIGSEGV, Segmentation fault.
ORB_SLAM2::KeyFrame::GetPose (this=this@entry=0x0) at /mnt/hdd/ORB_SLAM2/src/KeyFrame.cc:94
94	    if(!Tcw.empty()) {
(gdb) 

```

With the following backtrace:
```
(gdb) bt
#0  0x00007ffff7909916 in ORB_SLAM2::KeyFrame::GetPose() (this=this@entry=0x0) at /mnt/hdd/ORB_SLAM2/src/KeyFrame.cc:94
#1  0x00007ffff78be863 in ORB_SLAM2::Tracking::UpdateLastFrame() (this=this@entry=0x55555afbeac0)
    at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:820
#2  0x00007ffff78bf1c7 in ORB_SLAM2::Tracking::TrackWithMotionModel() (this=this@entry=0x55555afbeac0)
    at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:886
#3  0x00007ffff78c0b40 in ORB_SLAM2::Tracking::Track() (this=this@entry=0x55555afbeac0) at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:322
#4  0x00007ffff78c2241 in ORB_SLAM2::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&) (this=0x55555afbeac0, imRectLeft=..., imRectRight=..., timestamp=@0x7fffffffd7f8: 310.15600000000001) at /mnt/hdd/ORB_SLAM2/src/Tracking.cc:205
#5  0x00007ffff78aaf08 in ORB_SLAM2::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&) (this=0x7fffffffd990, imLeft=..., imRight=..., timestamp=@0x7fffffffd7f8: 310.15600000000001) at /mnt/hdd/ORB_SLAM2/src/System.cc:174
#6  0x0000555555557a79 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /mnt/hdd/ORB_SLAM2/Examples/Stereo/stereo_kitti.cc:100
```

Can anyone shed some light onto this peculiar subject?

Thanks :)

YumaTheCompanion, 2019-09-04:
Also, please see the valgrind output:

```valgrind
Arrived at GetPose().
==27042== Invalid read of size 8
==27042==    at 0x50D3916: empty (mat.hpp:404)
==27042==    by 0x50D3916: ORB_SLAM2::KeyFrame::GetPose() (KeyFrame.cc:94)
==27042==    by 0x5088862: ORB_SLAM2::Tracking::UpdateLastFrame() (Tracking.cc:820)
==27042==    by 0x50891C6: ORB_SLAM2::Tracking::TrackWithMotionModel() (Tracking.cc:886)
==27042==    by 0x508AB3F: ORB_SLAM2::Tracking::Track() (Tracking.cc:322)
==27042==    by 0x508C240: ORB_SLAM2::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&) (Tracking.cc:205)
==27042==    by 0x5074F07: ORB_SLAM2::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&) (System.cc:174)
==27042==    by 0x10BA78: main (stereo_kitti.cc:100)
==27042==  Address 0x3b0 is not stack'd, malloc'd or (recently) free'd
==27042== 
==27042== 
==27042== Process terminating with default action of signal 11 (SIGSEGV)
==27042==  Access not within mapped region at address 0x3B0
==27042==    at 0x50D3916: empty (mat.hpp:404)
==27042==    by 0x50D3916: ORB_SLAM2::KeyFrame::GetPose() (KeyFrame.cc:94)
==27042==    by 0x5088862: ORB_SLAM2::Tracking::UpdateLastFrame() (Tracking.cc:820)
==27042==    by 0x50891C6: ORB_SLAM2::Tracking::TrackWithMotionModel() (Tracking.cc:886)
==27042==    by 0x508AB3F: ORB_SLAM2::Tracking::Track() (Tracking.cc:322)
==27042==    by 0x508C240: ORB_SLAM2::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&) (Tracking.cc:205)
==27042==    by 0x5074F07: ORB_SLAM2::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&) (System.cc:174)
==27042==    by 0x10BA78: main (stereo_kitti.cc:100)
==27042==  If you believe this happened as a result of a stack
==27042==  overflow in your program's main thread (unlikely but
==27042==  possible), you can try to increase the size of the
==27042==  main thread stack using the --main-stacksize= flag.
==27042==  The main thread stack size used in this run was 8388608.
==27042== 
==27042== HEAP SUMMARY:
==27042==     in use at exit: 350,576,886 bytes in 1,236,139 blocks
==27042==   total heap usage: 2,853,492 allocs, 1,617,353 frees, 16,435,241,365 bytes allocated
==27042== 
==27042== LEAK SUMMARY:
==27042==    definitely lost: 22,680 bytes in 36 blocks
==27042==    indirectly lost: 155,959 bytes in 943 blocks
==27042==      possibly lost: 68,705,243 bytes in 1,083,087 blocks
==27042==    still reachable: 281,610,340 bytes in 151,399 blocks
==27042==                       of which reachable via heuristic:
==27042==                         length64           : 4,872 bytes in 81 blocks
==27042==                         newarray           : 20,616 bytes in 61 blocks
==27042==                         multipleinheritance: 1,808 bytes in 3 blocks
==27042==         suppressed: 0 bytes in 0 blocks
==27042== Rerun with --leak-check=full to see details of leaked memory
==27042== 
==27042== For counts of detected and suppressed errors, rerun with: -v
==27042== Use --track-origins=yes to see where uninitialised values come from
==27042== ERROR SUMMARY: 609443 errors from 3 contexts (suppressed: 0 from 0)

```

NCHUPJ, 2019-09-16:
I have encountered similar problems. Have you solved them?

YumaTheCompanion, 2019-09-16:
Yes, I've solved them. There seems to be an error on file: `void Tracking::StereoInitialization()`. I'll push my changes to my fork, and do a pull request on here.

barloccia, 2019-09-18:
About the first modify that you proposed, i am having problems understanding the behavior of ORBSLAM2 respect to different timestamps. I'm looking at the piece of code that you posted. If i increase the time between two taken images then the algorithm works well, if i decrease it often it lost track, indipendently from the fps. Time stamp is only used for sync? Can i change it as i want to fit the best performance? Thanks for the help

YumaTheCompanion, 2019-09-20:
As far as I can tell, timestamps are not even used during slam mode, they are there so as to provide the replay duration of the video correctly. So, if you choose to not provide a time stamp text and remove that portion of the code, the slam algorithm will work as fast as possible, and the resulting video will be played as fast as your computer can solve slam.

Other than that, my error would happen without regards to time stamps, further investigation revealed so. It would always segfault at frame 149, so that’s why I decided to look deeper into code. I’m not sure if it’ll fix your problems as well, you’ll have to try it out.

But the fix just changes the execution order of a small piece of code in the same function it resides, so I figured it wouldn’t affect the math as that part is only concerned with allocation and does not do maths.

YumaTheCompanion, 2019-09-20:
Also, since the error is a segfault, this means that software is trying to access a location on memory which doesn’t belong to him anymore. Changing time can cause this to happen on a faulty code. This is because the environment is multithreaded and a corruption is more likely to occur while you’re sleeping on another section

closed 2019-09-20 by YumaTheCompanion