raulmur/ORB_SLAM2 #812: RGB-D yaml: Camera.bf "IR projector baseline"

hoonkai, 2019-09-04:
Hi

I'm trying to find the value of `Camera.bf` for my RGBD camera. I believe the value is meant to be "IR projector baseline times fx (aprox.)", but how is the IR projector baseline measured? I presume `fx` here is the `fx` of the RGB camera, but aren't `fx`, `fy`, `cx` and `cy` of the IR camera need to be taken into account?

Any help will be much appreciated.

DimTrigkakis, 2019-09-11:
The IR projector is actually two projectors that provide a pattern of point lights for the sensor to pick up. You can find the distance between the two IR projectors either in the documentation or in some ros topic, through code etc.

Generally you would want to align the two images, the one from the depth and rgb camera, so that after aligning the frames you would only need to use the fx,fy,cx,cy of the unchanged camera stream.

collinabidi, 2019-11-02:
@DimTrigkakis How would you ask the realsense library for the camera intrinsics? I'm using the RealSense D435

DrChanging, 2020-06-17:
@collinabidi see this issue 
https://github.com/IntelRealSense/librealsense/issues/2580#issue-372949620

ph-code-repo, 2021-12-25:
@collinabidi If this is still an issue, you can do it through the command line with Intel.Realsense.CustomRW -sn ##-SerialNumber## -r -f ##FileName##
This was from the "Intel® RealSense™ Product Family D400 Series Calibration Tools" document - Chapter 6

animesh-singhal, 2024-01-26:
**Issue Description:**
- I sought information regarding the IR projector baseline for the D415 sensor on the [Framos product page](https://www.framos.com/en/products/depth-camera-d415-starter-kit-20745).
- Found information labeled simply as "baseline."

**Clarification Needed:**
- Is the listed "baseline" value equivalent to the IR projector baseline for the D415 sensor?
- What are the appropriate units for measuring the IR projector baseline in this context?