raulmur/ORB_SLAM2 #814: Centimeter level accuracy with D435

DimTrigkakis, 2019-09-11:
Hello,

I am trying to obtain very accurate results for the trajectory of a D435 camera. I move in a straight line while looking at the ground and the building on my left. The video stream is continuous and I have used both 640x480 and 1280x720 resolutions. 

I align the depth and color frames and run the RGB-D version of orb slam2. I get all settings from the camera (using it in the same resolution as each experiment) and produce the yaml file. After running orb-slam2, and viewing the point cloud, I see that at a distance of 15 meters the trajectory incorrectly estimates it at 15.90 meters.

In the paper where they evaluate results on the kitty dataset, I see that after loop closure for over 400 meters they get half a meter absolute translation error, which is significantly lower than what I see in my small experiment.

I have tried tinkering with the camera parameters (high-accuracy settings for depth map estimation), removing depth from outside a range [x, y] (for example only keep depth from 2 meters to 4) among other things.

I have run out of ideas, is the camera or orb-slam2 not good enough for accurate localization (centimeter-level) within a 20 meter range? Is it because I experiment in an outdoor environment? I have tried everything I could imagine, any help is appreciated.

thohemp, 2019-09-17:
What are your calibration settings?

DimTrigkakis, 2019-09-20:
A sample calibration file I generate in an automated way is the following:

`%YAML:1.0

Camera.fx: 613.889099121
Camera.fy: 613.900878906
Camera.cx: 324.221984863
Camera.cy: 245.42074585

Camera.k1: 0
Camera.k2: 0
Camera.p1: 0
Camera.p2: 0

Camera.width: 640
Camera.height: 480

Camera.fps : 30

Camera.bf : 30.6520567054
Camera.RGB : 0

ThDepth: 55
DepthMapFactor: 1000

ORBextractor.nFeatures: 1000
ORBextractor.scaleFactor: 1.2
ORBextractor.nLevels: 8
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500`

jwangjie, 2019-10-18:
@DimTrigkakis I was wondering can you share the calibration file you used for 1280x720 resolutions? Thank you.