raulmur/ORB_SLAM2 #821: Segmentation fault (core dumped)

95ImadEL, 2019-10-03:
Hello everyone.
I tried to run rgbd_tum program on rgbd_dataset_freiburg1_xyz dataset, but the program failed, it produced the following error:
```
imad@imad-FX503VM:~/Desktop/ORB_SLAM2/ORB_SLAM2$ ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml '/home/imad/Desktop/ORB_SLAM2/Dataset/rgbd_dataset_freiburg1_xyz' '/home/imad/Desktop/ORB_SLAM2/Dataset/associations.txt' 

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294

-------
Start processing sequence ...
Images in the sequence: 792

Segmentation fault (core dumped)
```

thank you in advance for your collaboration

jzhzhang, 2019-10-19:
@95ImadEL Hi, I meet the same problem with you. Have you found where is the mistake?

95ImadEL, 2019-10-19:
@jzhzhang  not yet unfortunately (on ubuntu 18), but when i run it on Ubuntu 14 it works.

jzhzhang, 2019-10-20:
@95ImadEL I change the opencv 3.2 to 3.4, and everything works perfectly.

skystalkerrc, 2020-05-22:
I have the same issue, Ubuntu 18.04, opencv 3.1

Miha22, 2022-05-04:
Build this OpenCV https://github.com/opencv/opencv/tree/2.4