raulmur/ORB_SLAM2 #822: compute pose covariance

arianna003, 2019-10-04:
Hi, how i can calculate the covariance of the estimated pose ?
thanks

AlejandroSilvestri, 2020-02-14:
@arianna003 
ORB-SLAM2 doesn't use covariance matrix.  It uses bundle adjustment - a whole different approach.

If you need to calculate it, you must do it yourself, starting with defining what variables you want to correlate.

mathuse, 2020-06-08:
> @arianna003
> ORB-SLAM2 doesn't use covariance matrix. It uses bundle adjustment - a whole different approach.
> 
> If you need to calculate it, you must do it yourself, starting with defining what variables you want to correlate.

I have the stereo camera, and I migrate the code to my code. I find some problem,the first problem is that what time the program calculate the "pKF->mFeatVec" when program first enter the "TrackReferenceKeyFrame/SearchByBoW". I slove it by adding "mCurrentFrame->ComputeBoW();" before "StereoInitialization/new KeyFrame" but it is not perfect, because I have successfully tested stereo camera using the executable file compiled from ORB-SLAM2 code.
The second problem is "std::bad_alloc" when I run the my executing file, and I have find the location of this problem at "TrackReferenceKeyFrame/PoseOptimization/optimizer.optimize(its[it])". I want to know the reason.Can you help me? By the way, I have tried a lot of ways, but none of them worked.