raulmur/ORB_SLAM2 #827: symbol lookup error: libORB_SLAM2.so, undefined symbol: _ZN5DBoW24FORB1LE

HayatoTajiri, 2019-10-15:
environment(環境)

||
|-|
|Jetson nano| 
|Ubuntu 18.04LTS|
|ROS melodic|
|opencv ver.3.3.1|

I had used orb_slam2 on jetson nano with Ros melodic.
However I haven't been able to use Orb_slam2 before installing Realsense r200 drivers(librealsense,realsense_camera,etc). 

error content(エラー内容)

Using with Ros
`
$ rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml
/opt/ros/melodic/share/ORB_SLAM2/RGBD: symbol lookup error: /usr/local/lib/libORB_SLAM2.so: undefined symbol: _ZN5DBoW24FORB1LE
` 
Using without Ros
`
~/orb_slam2_ws/ORB_SLAM2/Examples/Monocular/mono_tum Vo
cabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml
/home/hayato/orb_slam2_ws/ORB_SLAM2/Examples/Monocular/mono_tum: symbol lookup error: /usr/local/lib/libORB_SLAM2.so: undefined symbol: _ZN5DBoW24FORB1LE
`

I can't understand what causes.
please tell me solutions,anyone.
thanks.

mathuse, 2019-11-08:
> ## I can't run orb_slam2.
> _environment(環境)_
> Jetson nano
> Ubuntu 18.04LTS
> ROS melodic
> opencv ver.3.3.1
> I had used orb_slam2 on jetson nano with Ros melodic.
> However I haven't been able to use Orb_slam2 before installing Realsense r200 drivers(librealsense,realsense_camera,etc). Probably.
> 
> error content(エラー内容)
> 
> Using with Ros
> `$ rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml /opt/ros/melodic/share/ORB_SLAM2/RGBD: symbol lookup error: /home/hayato/orb_slam2_ws/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../lib/libORB_SLAM2.so: undefined symbol: _ZN5DBoW24FORB1LE`
> Using without Ros
> `$ ~/orb_slam2_ws/ORB_SLAM2/Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml /home/hayato/orb_slam2_ws/ORB_SLAM2/Examples/Monocular/mono_tum: symbol lookup error: /home/hayato/orb_slam2_ws/ORB_SLAM2/lib/libORB_SLAM2.so: undefined symbol: _ZN5DBoW24FORB1LE`
> 
> I can't understand what causes.
> please tell me solutions,anyone.
> thanks.

hi,i have the same problem now,and did you have solved it?

FaikVural, 2020-03-09:
I have same problem, Please help us

HayatoTajiri, 2020-03-09:
Hi.
Finally, I could't solve it but, I 
formated SDcard and reinstalled ubuntu, so it is running.

SkylerGao, 2020-09-11:
I also have the same problem, maybe the version problem of DBOW ? Waiting for the solution from "Big God (Chinese words)"...

SkylerGao, 2020-09-11:
forgot to say... I met the problem in ORB-SLAM3

segozcan, 2020-09-15:
I encountered the same issue with ORB_SLAM2. What solved the issue for me moving ORB_SLAM2 to a different directory whch had not been built by catkin build. In my case, I had tried to add new packages to the workspace and build them with catkin_build. Undoing that solved the problem for me.

wh200720041, 2021-03-14:
Hi this is caused by repetitive installation of DBoW2 in my case. I searched for the DBoW2 in my computer and removed all files named *DBoW2* under /usr/local and the problem was solved

Viky397, 2021-03-16:
Hi, I've tried that @wh200720041 and the error still persists for me. Any other recommendations? Thanks!

wh200720041, 2021-03-17:
> Hi, I've tried that @wh200720041 and the error still persists for me. Any other recommendations? Thanks!

Make sure all previous DBoW is removed, including DBoW folder and its related files under /usr/.

dnaglucose, 2021-05-22:
`ldd -r libORB_SLAM3.so`
find:
`libDBoW2.so => xxxxx/xxxxx/xxxxx/lib/libDBoW2.so`
remove this libDBoW2.so and make again, can solved this problem

https://blog.csdn.net/qq_35942419/article/details/117163042

Shivelino, 2022-08-12:
> Hi this is caused by repetitive installation of DBoW2 in my case. I searched for the DBoW2 in my computer and removed all files named _DBoW2_ under /usr/local and the problem was solved

correct, DBoW2 is one of thirdparties, but I have installed one before

mathuse, 2022-08-12:
Thank you very much.

tigerschweiger, 2024-01-09:
> Hi this is caused by repetitive installation of DBoW2 in my case. I searched for the DBoW2 in my computer and removed all files named _DBoW2_ under /usr/local and the problem was solved

It works! Thx a lot~