raulmur/ORB_SLAM2 #828: How to run ros Stereo node with kitti?

AndroidStudio2017, 2019-10-16:
Hi, I'm stuck in another problem.

I tried to run Stereo node with kitti, so I run the command as follow:
`rosrun ORB_SLAM2 Stereo /home/johnson/ORB_ws/src/ORB_SLAM2-master/Vocabulary/ORBvoc.txt /home/johnson/ORB_ws/src/ORB_SLAM2-master/Examples/Stereo/KITTI00-02.yaml false`

it works, but when I use command to play rosbag, SLAM don't work.
the command I use to play rosbag:
`rosbag play --pause ~/Dataset/kitti/kitti00_stereo_30fps.bag /image_left:=/camera/left/image_raw /image_right:=/camera/right/image_raw`

The following is the command info :

Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 0 points
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 0 points

I don't know why that happens. Thanks for any answer.

mhaboali, 2020-07-01:
Same issue here with my graduation project, could you help?

withnoerr, 2024-07-12:
> 我的毕业项目也有同样的问题，你能帮忙吗？

Did you solve that? I have the same problem.