raulmur/ORB_SLAM2 #833: How to get correct YAML parameters

collinabidi, 2019-10-31:
I'm currently using a RealSense 435 RGB-D camera to collect data that I feed into ORB_SLAM's `rgbd_tum` example. However, the parameter file confuses me.

I use a modified version of the TUM_1.yaml file, which is as follows:

`%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# Camera calibration and distortion parameters (OpenCV) 
# K ->  | fx  0  width/2   |
#       | 0  fx  height/2  | 
#       | 0   0    1.0     |
# 
# P ->  | fx  0  width/2   0  |
#       | 0  fx  height/2  0  | 
#       | 0   0    1.0     0  |

# INTEL REALSENSE FOV
# VGA 4:3  ->  horiz = 74, vertical = 62
# HD 16:9  ->  horiz = 86, vertical = 57
# IR BASELINE: 49.

Camera.fx: 686.315974421
Camera.fy: 386.052735612
Camera.cx: 640 # width / 2
Camera.cy: 360 # height / 2

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 30.0

# IR projector baseline times fx (aprox.)
Camera.bf: 49.8

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 100

# Depthmap values factor 
DepthMapFactor: 10
#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 2000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 10

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

`

The parameters I'm confused by are ThDepth, DepthMapFactor, and all of the ORB Parameters. I have found optimal ThDepth and DepthMapFactor through trial and error, but what works for one video doesn't work for another.

Good result:

[image]

Bad result:

[image]

The two videos I use are taken with the same camera in the same space with the same settings.

Is there a way to figure out what parameters will always work for the same camera?

ishanivyas, 2019-11-16:
@collinabidi   Could you explain me how do you calculate# Close/Far threshold. Baseline times.
ThDepth: 10

DimTrigkakis, 2019-11-21:
The IR baseline should be in meters, so it's 0.050 (the D435 has documentation that mentions 50mm not 49). 

Also, the depthmapfactor is the conversion factor between depth values and real depth. In the D435 it's 1000 (1000 inside the depth map corresponds to a meter)

Dennis2Choi, 2019-11-28:
@collinabidi @DimTrigkakis 
Document says "# IR projector baseline times fx (aprox.)" 
but I think you both are talking about "IR baseline" for the left/right image sensor. 
Is it same ? and is there any meaning for "times fx" ? I wonder 50 or 49.8 is correct one...

DimTrigkakis, 2019-11-28:
I changed 0.05 (5cm) to 0.03 (3cm) for the proper baseline from the IR projector to the left IR sensor center (which is the origin of the IR-depth coordinate system) and run two experiments. I couldn't see any major differences in the result, but there was definitely no "bad result" as shown in the images above.