raulmur/ORB_SLAM2 #835: load/save map and relocalization

buemi, 2019-11-04:
Hi all.
I used the fork by Alkaid-Benetnash enabling the possibility to save and load a map at the beginning as described here:

https://github.com/Alkaid-Benetnash/ORB_SLAM2

I also added the possibility to acquire data in streaming from a web-cam (moving it by hands).
I was expected that the relocalization block works at any time, since loading the map transforms the environment from unknown to known. But it doesn't seem the case. The algorithm seems to work as the original one: when tracking is lost, relocation works only by going back a few steps. 
Is there something wrong in my assumptions?

buemi, 2019-11-05:
Sorry, my fault: the compile process doesn't overwritten the old executable. 
"Make clean" before build again solved the problem.

closed 2019-11-05 by buemi

Shame-fight, 2020-03-13:
> Sorry, my fault: the compile process doesn't overwritten the old executable.
> "Make clean" before build again solved the problem.

Hello, I just compiled the code for Alkaid-Benetnash, but I have some doubts about saving and loading maps with monocular and binocular cameras, because they run on ros. Can you tell me how I should modify the code? Looking forward to hearing from you

buemi, 2020-03-13:
Hi!
Well, I just followed the readme section 11. instructions and I modified accordingly the yaml, mono and system.cc files adding the statements "SLAM.SaveMap(mapfilename)" and LoadMap(MapFile):

11. Map Save/Load

Enable:

Considering this feature doesn't hurt performance, and it is annonying to deal with conditional compilation flags, so this feature will be enabled unconditionally.

Usage:

This feature is integrated with class System. The path of mapfile can be set by adding Map.mapfile: map.bin to ORB_SLAM2's settings file. See the last few line of TUM1.yaml for example.

To save a map, you need construct ORB_SLAM2::System with the last parameter (is_save_map) be true. Then the System will save map to mapfile (create if non-existent, overwrite if existent) specified in then setting file when ShutDown (e.g. interrupted by ctrl+c).

With a readable mapfile, map will be loaded automatically and System will run in localization mode at first, but you can change it to SLAM mode later.

mono_tum has been updated as a simple example of this functionality. An extra command line parameter(0 or 1) should be given to indicate whether you want to save map or not.

Implementation related:

I use boost_serialization library to serialize Map, MapPoint, KeyFrame,KeyFrameDatabase, cv::Mat, DBoW2::BowVector, DBoW2::FeatureVector. In brief, only the ORBVector isn't serialized.

This feature is tested with boost 1.64 and it works fine mostly. There is still some occasional segmentfault to dig in.

Shame-fight, 2020-03-13:
> Hi!
> Well, I just followed the readme section 11. instructions and I modified accordingly the yaml, mono and system.cc files adding the statements "SLAM.SaveMap(mapfilename)" and LoadMap(MapFile):
> 
> 1. Map Save/Load
> 
> Enable:
> 
> Considering this feature doesn't hurt performance, and it is annonying to deal with conditional compilation flags, so this feature will be enabled unconditionally.
> 
> Usage:
> 
> This feature is integrated with class System. The path of mapfile can be set by adding Map.mapfile: map.bin to ORB_SLAM2's settings file. See the last few line of TUM1.yaml for example.
> 
> To save a map, you need construct ORB_SLAM2::System with the last parameter (is_save_map) be true. Then the System will save map to mapfile (create if non-existent, overwrite if existent) specified in then setting file when ShutDown (e.g. interrupted by ctrl+c).
> 
> With a readable mapfile, map will be loaded automatically and System will run in localization mode at first, but you can change it to SLAM mode later.
> 
> mono_tum has been updated as a simple example of this functionality. An extra command line parameter(0 or 1) should be given to indicate whether you want to save map or not.
> 
> Implementation related:
> 
> I use boost_serialization library to serialize Map, MapPoint, KeyFrame,KeyFrameDatabase, cv::Mat, DBoW2::BowVector, DBoW2::FeatureVector. In brief, only the ORBVector isn't serialized.
> 
> This feature is tested with boost 1.64 and it works fine mostly. There is still some occasional segmentfault to dig in.

Thanks for your reply, I'm sorry I still can't understand. Can you point out the specific location (such as the number of lines) and the modified code for each file. I apologize for the inconvenience.

Shame-fight, 2020-03-14:
> Hi!
> Well, I just followed the readme section 11. instructions and I modified accordingly the yaml, mono and system.cc files adding the statements "SLAM.SaveMap(mapfilename)" and LoadMap(MapFile): [...]

@buemi  
Whether you can provide the specific modification location or the modified file. Sorry I just got in touch with SLAM, I have been stuck for a day and I look forward to hearing from you.

buemi, 2020-03-15:
Hi. Sorry, I was off-line for a while.
Well, if I remember well, you did all the needed steps. Just rebuild using the build.sh script (after 'make clean' that you have to launch from terminal in the build folder. You also can simply rename build folder).  Note that I didn't tested the ros version.