raulmur/ORB_SLAM2 #836: /usr/bin/ld: warning libopencv_core.so.2.4 needed by ... may conflict with libopencv_core.so.3.4

witignite, 2019-11-08:
## Error:
```
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target stereo_kitti
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target mono_kitti
[100%] Built target rgbd_tum
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target mono_tum
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target mono_euroc
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
/usr/bin/ld: warning: libopencv_imgproc.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_imgproc.so.3.4
[100%] Built target stereo_euroc
```

## **How I got this error:**

I have my `OpenCV 3.4.7` built with `FFMPEG` (if it is relevant) and installed to a custom path in the `/opt/` folder. And so I modify the `ORB_SLAM2`'s build script `build.sh` by passing the `OpenCV_DIR` to `cmake` so that it can find my `OpenCV 3.4.7` as follows:
```
...
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DOpenCV_DIR=/opt/OpenCV/opencv-3.4.7/share/OpenCV
make -j
```
When I execute the build script, here is the output:
```bash
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-10.0 (found suitable exact version "10.0") 
-- Found OpenCV: /opt/OpenCV/opencv-3.4.7 (found suitable version "3.4.7", minimum required is "3.0") 
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.1.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ORB_SLAM2/build
Scanning dependencies of target ORB_SLAM2
[  6%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
[  6%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
[  9%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/ORBmatcher.cc.o
[ 12%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
[ 15%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o
[ 18%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o
[ 21%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/ORBextractor.cc.o
[ 25%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrame.cc.o
[ 31%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o
[ 31%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Sim3Solver.cc.o
[ 37%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o
[ 40%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o
[ 43%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrameDatabase.cc.o
[ 46%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o
[ 50%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Map.cc.o
[ 56%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o
[ 56%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
[ 59%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
[ 62%] Linking CXX shared library ../lib/libORB_SLAM2.so
[ 62%] Built target ORB_SLAM2
Scanning dependencies of target mono_tum
Scanning dependencies of target mono_kitti
Scanning dependencies of target mono_euroc
Scanning dependencies of target rgbd_tum
Scanning dependencies of target stereo_kitti
Scanning dependencies of target stereo_euroc
[ 65%] Building CXX object CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o
[ 68%] Building CXX object CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o
[ 71%] Building CXX object CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o
[ 78%] Building CXX object CMakeFiles/stereo_euroc.dir/Examples/Stereo/stereo_euroc.cc.o
[ 78%] Building CXX object CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o
[ 81%] Building CXX object CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o
[ 84%] Linking CXX executable ../Examples/Monocular/mono_kitti
[ 87%] Linking CXX executable ../Examples/Stereo/stereo_kitti
[ 90%] Linking CXX executable ../Examples/RGB-D/rgbd_tum
[ 93%] Linking CXX executable ../Examples/Monocular/mono_euroc
[ 96%] Linking CXX executable ../Examples/Monocular/mono_tum
[100%] Linking CXX executable ../Examples/Stereo/stereo_euroc
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target stereo_kitti
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target mono_kitti
[100%] Built target rgbd_tum
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target mono_tum
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
[100%] Built target mono_euroc
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.4
/usr/bin/ld: warning: libopencv_imgproc.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_imgproc.so.3.4
[100%] Built target stereo_euroc
```

annonp, 2019-11-08:
I have exactly the same problem and the same error on my Ubuntu 16.04:
- It looks like **`libavfilter-ffmpeg.so.5`** requires **`libopencv_core.so.2.4`** library, which somehow is already installed in **`/usr/lib/x86_64-linux-gnu/`**.
- At the same time,  **`libopencv_core.so.3.4`** (*built from source*) is given to the linker `ld`, and this causes a conflict.

So I tired to remove `libopencv*` by:
```bash
sudo apt remove libopencv*
...
The following packages will be REMOVED
  ffmpeg libavdevice-dev libavdevice-ffmpeg56 libavfilter-dev libavfilter-ffmpeg5
  libopencv-core2.4v5 libopencv-imgproc2.4v5
```
But this gave me error when builiding `ORB_SLAM2` again:
```bash
...
[ 31%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libavdevice.so', needed by '../lib/libORB_SLAM2.so'. Stop.
make[2]: *** Waiting for unfinished jobs....
```

witignite, 2019-11-10:
For some reason I still don't know, it seems that the build of my `OpenCV 3.4.7` was not correct. So I reinstalled my Ubuntu, starting with a clean system, re-building the `OpenCV` from source. And this time, it works.
For my curiosity, I tried to search for `libavfilter` and `ffmpeg` and other libraries that had caused this error previously, but they are still right there.
```bash
user@laptop:~$ apt list --installed | grep ffmpeg

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

chromium-codecs-ffmpeg-extra/xenial-updates,xenial-security,now 78.0.3904.70-0ubuntu0.16.04.2 amd64 [installed,automatic]
libavcodec-ffmpeg56/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libavfilter-ffmpeg5/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libavformat-ffmpeg56/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libavresample-ffmpeg2/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libavutil-ffmpeg54/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libpostproc-ffmpeg53/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libswresample-ffmpeg1/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
libswscale-ffmpeg3/xenial-updates,xenial-security,now 7:2.8.15-0ubuntu0.16.04.1 amd64 [installed,automatic]
```
The `libopencv*` are still right there as well:
```bash
w@laptop:~$ apt list --installed | grep libopencv

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libopencv-calib3d2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-contrib2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-core2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-features2d2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-flann2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-highgui2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-imgproc2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-legacy2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-ml2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-objdetect2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
libopencv-video2.4v5/xenial-updates,xenial-security,now 2.4.9.1+dfsg-1.5ubuntu1.1 amd64 [installed,automatic]
```
I'm just curious why it doesn't cause library conflicts like before...

closed 2019-11-28 by witignite

rancheng, 2020-01-05:
you installed the libavfilter-ffmpeg from apt before you build opencv.