raulmur/ORB_SLAM2 #839: The unit of distance in depth map?: Can not see the key points in map when using my own RGB-D sequence

MRinel, 2019-11-14:
Hello everyone,
I use my own RGB-D squence to test ORB_SALM2 but cannot see the key points in display.
**First**,RGB-D images were from TUM RGB , predicted by network and output in meters. (I have no RGB-D camera )
Here is the depth matrix:
[image]

**Second**, after normalization for saving, the shape of RGB-D was [224 ,224, 4]. 
[image]

Then, using the RGB-D routine, but I couldn't see the key points. It seems that there is something wrong with the distance because the key points are jammed together... like:
[image]

In author's example ,the shape of TUM RGB-D is [640,480], I wonder that what is the unit of  distance in RGB-D,meter or millimeter feed in SLAM system? and ,any device for my problem? Thanks a lot !!
Best wishes.

collinabidi, 2019-11-15:
I'm having the same issue.

MRinel, 2019-11-18:
@collinabidi , Is it getting too many feature points? Or you can scroll in or out to see it.
For me, after adjusting the factors,**"DepthMapFactor"**and **"ThDepth"** , results are slightly better...like:
[image]
There is still much work to be done for using my own video.
By the way, I am not clear with **"DepthMapFactor"**...and this part in **Tracking.cc**:
[image]
[image]
I would appreciate it if anyone could explain it, Thanks a lot! 
 
Best Wishes :)

collinabidi, 2019-11-18:
@MRinel For this particular map it does get way too many feature points. I am taking a video of a highly cluttered environment though, so zooming out doesn't really help much anyways (in my case).

I'm not exactly clear what that means either. I know that when I record my video, I can set my depth magnitude value to whatever value I want, but it seems like trial and error was the only thing that helped me find better values.

MRinel, 2019-11-19:
@collinabidi ,My guess is that "DepthMapFactor" may represent the scaling factor between true scale and displaying in the map, because the origin value in authors' is 5000.0 ,(maybe the unit of true distance is millimeter,mine is in meters), and when I set it to 4, these feature points can be found very well. So, "DepthMapFactor" is just a factor to display a map better. We need to keep trying to find a more proper value.

closed 2019-11-20 by MRinel

collinabidi, 2019-11-20:
@MRinel I found that 4 seemed to work for me as well, not sure why that was good.

Mechazo11, 2024-05-24:
@collinabidi Hi I know this is a old post, but are you familiar with a technique of finding the DepthMapFactor itself for a camera not previously used in ORB-SLAM3? For instance, I am trying to use the ORBEC ASTRA S camera and so far I haven't found an exact method of determining this value