raulmur/ORB_SLAM2 #841: Best way to get the feature/key points from each frame

Aeogor, 2019-11-22:
Hello, I am trying to create a mask based on the feature points generated by the SLAM monocular from the previous frame and apply the mask to the current frame. However, I am see multiple ways of getting these features and am unsure which is accurate way. 

`Ptr<Feature2D> detector = ORB::create(MAX_FEATURES);
 std::vector<KeyPoint> keypoints;
 cv::Mat descriptors;
 detector->detectAndCompute(im, cv::Mat(), keypoints, descriptors);`

I am able to get the feature points thorugh this method, however the points key points generated are diffferent from this method. 

`std::vector<KeyPoint> keypoints = SLAM.GetTrackedKeyPointsUn(); `

The function GetTrackedKeyPointsUn returns a lot more points than the ORB detector method. What is the difference in these two approaches? And what exactly is GetTrackedKeyPointsUn returning? 

Thank you.

flo981, 2019-12-03:
Hey,

In System.cc
```
vector<cv::KeyPoint> System::GetTrackedKeyPointsUn()
{
    unique_lock<mutex> lock(mMutexState);
    return mTrackedKeyPointsUn;
}
```
and 
`mTrackedKeyPointsUn = mpTracker->mCurrentFrame.mvKeysUn;`

mTrackedKeyPointsUn is a vector including the currently tracked undistorted keypoints.

But how do you use your first method? I can't find any of these functions in ORBSLAM2.

AlejandroSilvestri, 2019-12-06:
@Aeogor 

SLAM.GetTrackedKeyPointsUn() will get you the keypoints detected in the current frame in ORB_SLAM2.

ORB_SLAM2 applies an ad hoc ORB implementation.  It doesn't use OpenCV ORB detector.  This is what ORB_SLAM2 does different than regular ORB:

- Divide the image in many little images, and apply FAST with a threshold on them.
- On each little image with few keypoints, re run FAST with a more generous threshold.  This results in a more evenly distribution of keypoints on the whole image.

That's why the number of keypoints differs on those got with ORB detector, no matter what threshold you use.

It continues with the same steps ORB does: on each keypoint computes angle and then extracts rotated descriptor.

closed 2019-12-09 by Aeogor

mmahdavian, 2020-10-14:
@Aeogor @AlejandroSilvestri 
Hi. Iam using SLAM.GetTrackedKeyPointsUn() to get keypoints in ros_mono.cc code. Do you how can I get descriptors for each feature point? I need to publish detected features to do feature matching on another code.

Thanks

AlejandroSilvestri, 2020-10-14:
@mmahdavian 

SLAM is an instance of System class.  There's no System's method to do that.

`System.GetTrackedKeyPointsUn()` gets the keypoints from `mpTracker->mCurrentFrame.mvKeysUn`

You can get descriptors from `mpTracker->mCurrentFrame.mDescriptors`.  But being `System.mpTracker` private, you should change it to public.

`mDescriptors` is a Mat with `mCurrentFrame.N` rows, one row per descriptor, being N the size of `mvKeysUn`.