raulmur/ORB_SLAM2 #850: RGB-D with Azure Kinect

Dennis2Choi, 2019-12-24:
Is there anyone who tries to work with Azure Kinect (New released one)

I'm working with D415 and Azure Kinect 

but it is not working well with Azure Kinect 

Any help or hint about setting yaml file on Azure Kinect ?

 - Both Camera data resized with 640x480 
 - Azure Kinect using NFOV 
 - Using RGB-D ORB-SLAM2 
 - DO I NEED TO CHANGE something with ./example/rgbd_tum ?
 - wonder what I need to put it with baseline * fx (can't understand fully about this value )

Happy holidays... 
Thanks in advance

magnusbarata, 2019-12-25:
Have you check this issue by @nonlinear1 on [Kinect Azure's github](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/725#)? It seems like for high-res rgbd input, orbslam2 tracking could get lost easily.

Or maybe you could try using [Kinect Azure ros driver](https://github.com/microsoft/Azure_Kinect_ROS_Driver) to publish rgbd images from Kinect Azure node in ROS, and feed it to orbslam2 ROS node.

I am currently trying both approaches, but none seems to work. Maybe we could help each other.

For the camera intrinsic values such as fx, fy, you could try using kinect Azure's `k4a_device_get_calibration()` function.

nonlinear1, 2019-12-25:
I think you should undistort your picture! I give the issue, and finally, I solve this through undistort the picture!!!

PierrePlantard, 2020-04-26:
Dear @nonlinear1,
I'm trying to get good results with kinect azure but without success, do you have a sample code that work with kinect Azure or a step by step explanation of the process ?
Best regards

nonlinear1, 2020-04-27:
Give me your e-mail. I will send you the code for input of kinect for azure!

PierrePlantard, 2020-04-27:
@nonlinear1 thank you.
My email is: pierre.plantard88@gmail.com

minsoo5403, 2020-06-03:
@ nonlinear1 Sir, can I have that?, please?
My email : minsoo5403@gmail.com

minsoo5403, 2020-06-03:
> Give me your e-mail. I will send you the code for input of kinect for azure!

Sir, can I have that?, please?
My email : minsoo5403@gmail.com

zhanghua7099, 2021-01-29:
@nonlinear1 
Can you give me the code for the input for azure Kinect?
I have run ORB-SLAM2 in offline datasets captured with 720P, but the running speed is too slow.
I guess that it may be some errors with the parameter setting.

My email: huayan.zhang@hotmail.com
Thank you!

ChenYll, 2022-03-05:
> 

15684165359@163.com
Could you please send me the code, I found the same problem, thanks!

CJAPPLE5, 2024-01-02:
@nonlinear1 
2081531631@qq.com
Could you please send me the code,, thank you very much!

CooRia, 2025-01-30:
@nonlinear1 
2306767421@qq.com
Could you please send me the code, i run RGBD but the tracking and mapping is poor. Thanks!