raulmur/ORB_SLAM2 #854: Spanning Tree Essential Graph

daothanlong11, 2020-01-05:
Hi every one, can some one help me understand the purpose of the spanning tree, what is its function. I had read ORB-SLAM2 and ORB-SLAM paper, i can understand what covisibility nd essential is doing but not the spanning tree, what part in the code use the spanning tree. Thank for your help

AlejandroSilvestri, 2020-02-12:
@daothanlong11 

If I remember right, the spanning tree is only used in loop closing, to evenly distribute the "error" in keyframes poses when fusing two ends to form a loop.

And it provides an convenient way to traverse the whole map if you need it.

yiakwy, 2020-03-05:
@daothanlong11 spanning tree may also make the pose graph is end circle path free in navigation. @daothanlong11 could you help me on this question https://github.com/raulmur/ORB_SLAM2/issues/872