raulmur/ORB_SLAM2 #857: Gray-scale image instead of color

lokeshluckybisht, 2020-01-26:
Hello,

I am facing an issue where while running the mono or stereo node through ROS, I am getting a gray-scale image in tracking display instead of color.

I am sending BGR8 packed image format from my camera. When I display the image after using OpenCV, the images are colored. The camera used is Baumer VLG-20C.I 

I know it must check for 3 channels but I am unable to understand why is it not detecting 3 channels.

Any help will be appreciated.

AlejandroSilvestri, 2020-02-12:
@lokeshluckybisht 

ORB-SLAM2 uses FAST and BRIEF, which work on grayscale.  Your color images are converted to grayscale as the very first step.  Viewer shows orb-slam2 input image, not the one you supply.

lokeshluckybisht, 2020-02-12:
> @lokeshluckybisht
> 
> ORB-SLAM2 uses FAST and BRIEF, which work on grayscale. Your color images are converted to grayscale as the very first step. Viewer shows orb-slam2 input image, not the one you supply.

Thank you !!

closed 2020-02-12 by lokeshluckybisht

bhaveshachhada, 2020-03-16:
Hello @AlejandroSilvestri , I wanted to know if depth information is being used while relocalization ?
I know that the code is somewhat different for RGBD and Mono upto creation of a frame, but afterwards, the code is same for both type of images. I wanted to know that If I have created and saved a map with RGBD camera, is it possible to relocalize with images taken from Mono RGB cam ? ( when I load the map back, it doesn't have any information wether the loaded map was created using a Mono, stereo or RGBD , then can I localize a RGB image ? )

AlejandroSilvestri, 2020-03-16:
@bhaveshachhada 

Well, ORB-SLAM2 doesn't load maps, so it highly depends on what library are you using.

From my knowledge on monocular, I strongly believe the first part in relocalization works the same way: it uses the inverted database of BoW to quickly recover the most likely keyframes, those similar to current frame.  Then there is a pose fitting that can be different.  But the info required from the map must be the same in both cases, RGBD and monocular.

So, I believe you can localize a RGB image on a stereo or RGBD based map.

bhaveshachhada, 2020-03-17:
Thank You @AlejandroSilvestri  Sir.