raulmur/ORB_SLAM2 #859: Passing pose data into ORBSLAM

Kiruko, 2020-01-28:
Hi everyone !
It might sound really weird but I would like to override (mostly) the localization system with my own pose data that is taken from my tracking device that gives me the pose of my camera.To sum up my problem, I want to use ORBSLAM to map the environment and also to take over in case the tracking is lost by my tracking device. Does anyone have any pointers as to where to start ? I read that maybe I should apply a Kalman filter or assign it randomly to certain frame.

Thanks in advance

AlejandroSilvestri, 2020-02-12:
@Kiruko 

System's `mpTracker->mCurrentFrame.SetPose(Tcw)` will do.

Tcw is a 4x4 float Mat with the World pose in reference to the Camera.

Kiruko, 2020-05-04:
Thank you for the reply. However, this didn't work for me. I succeeded in getting a good initialization of SLAM with the correct orientation and position by modifying it to triangulate keypoints between two frames as we know the position of the camera for those instead of using ORBSLAM to calculate them. 
Now, I would like to get a good accuracy of the camera pose after that initialization. However, simply passing the pose data doesn't seem to work (maybe too noisy). Because of that, I considered simply readjusting the position of the camera if the difference between ORB SLAM results and my tracking data is too great.
However, the motion model gives me a result that is very different from the data from my tracking device which results in a huge drift. The mapping however is still correct so I think that it is a scale problem. I noticed for example that mVelocity was very different compared to the one calculated using my tracking data.
Can someone help me understand why the motion model gives me such different results so that I can understand what I might have missed ?

NJNischal, 2023-04-03:
Hello @Bich-Anh  were you able to fix this issue of passing the ground truth camera pose to the ORBSLAM algorithm? If so, could you please explain what changes you have made to make it work? Thank you!