raulmur/ORB_SLAM2 #870: OpenCV 4 support

Norfo, 2020-02-27:
Hello everyone!
Are there any plans to support opencv 4? There is no big changes of code. Only in one place I have difficulties with the transition. The PnPsolver class is implemented with C Api. So I cann't find working solution for it. I replaced some functions and objects to support opencv 4. Everything seems to work well. But with the loss of tracking, he can not recover. I understand that maybe I need to contact with creators of EPnP. But maybe someone already faced with this problem and can anybody help me?

IaroslavS, 2020-02-29:
> Hello everyone!
> Are there any plans to support opencv 4? There is no big changes of code. Only in one place I have difficulties with the transition. The PnPsolver class is implemented with C Api. So I cann't find working solution for it. I replaced some functions and objects to support opencv 4. Everything seems to work well. But with the loss of tracking, it can not recover. I understand that maybe I need to contact with creators of EPnP. But maybe someone already faced with this problem and can anybody help me?

Hello. Could you kindly tell, what have you changed to work with opencv 4 ?

Norfo, 2020-03-02:
Good day, @IaroslavS! Firstly I've changed dependencies to opencv4. Then I replaced the code, which is not compatible with the new version. These replacements were mainly like `cvtColor(imGrayRight, imGrayRight, CV_RGB2GRAY);` -> `cvtColor (mImGray, mImGray, cv :: COLOR_RGB2GRAY) ;`. As I wrote above, the main problem was  to make changes in the PnpSolver file. There were many replacements in it and I'm not sure that I made them correctly. I've attached this class.
[PnPSolver.zip](https://github.com/raulmur/ORB_SLAM2/files/4273828/PnPSolver.zip)

In addition, I tried to inject code from another project (https://github.com/slowplayer/epnp-eigen). In it, the PnPSolver code use Eigen functionality. However, here I also could not get good results, because If tracking is lost, a critical error occurs.

Norfo, 2020-03-03:
The problem with porting to opencv 4 has been resolved. If somebody intresting, I could share my solution here.

closed 2020-03-03 by Norfo

Cenbylin, 2020-04-26:
Hi @Norfo , thanks for your great work. I would like to ask you if your attached code (PnPSolver.zip)  can work correctly? If not, could you share the new solution here?

Norfo, 2020-06-02:
@Cenbylin, sorry to have to wait. The decision was very trivial. Opencv 4 already has a solvepnp epnp method. Therefore, you do not need the main functionality of the pnpsolver class. This is its code
[PnPsolver.zip](https://github.com/raulmur/ORB_SLAM2/files/4715941/PnPsolver.zip). There are some debug console output. I think you can erase they if it necessary.

Cenbylin, 2020-06-02:
@Norfo  Great! Thanks.

zhuangxiaopi, 2020-09-25:
could you provide full project ? thx

Willforcv, 2021-12-07:
> The problem with porting to opencv 4 has been resolved. If somebody intresting, I could share my solution here.

did you use which version of ubuntu？